openrocket/openrocket #1821: [Personal annoyance] Replace "out of date" and "up to date" with "out-of-date" and "up-to-date"

SiboVG, 2022-11-13:
See title ⬆️ This concerns the hover text of simulation statuses (and maybe others).

JoePfeiffer, 2022-11-13:
For what it's worth, I strongly prefer it as is (with spaces) over the suggested change (with hyphens).

SiboVG, 2022-11-13:
> For what it's worth, I strongly prefer it as is (with spaces) over the suggested change (with hyphens).

Well I'll ask feedback from a native English speaker. I know there is a linguistic difference between "out of date" and "out-of-date". "out of date" is an adverb, while "out-of-date" is an adjective. OpenRocket uses "Data is out of date", and simply "Up to date" (which I would follow with 'data', so "Up to date data"). Aren't both "out of date" and "up to date" adjectives in this case? Sure, "out of date" correlates to "is", but at least in Dutch, adverbs related to "to be" are often just considered as adjectives.

I may be pushing this personal annoyance too far, and I'm happy to trust the gut feeling of native speakers.

JoePfeiffer, 2022-11-13:
My preference is simply a gut reaction.  It will be interesting to see if the other native English speakers agree with me.

I hadn't even noticed OR used "Data is out of date".  Whatever else happens, the "up to date" and "out of date" hover text should be consistent with each other; here I have a very weak preference for not including "data is", just because it's shorter and I think it's equally clear that way.

SiboVG, 2022-11-14:
> I hadn't even noticed OR used "Data is out of date". Whatever else happens, the "up to date" and "out of date" hover text should be consistent with each other; here I have a very weak preference for not including "data is", just because it's shorter and I think it's equally clear that way.

Agreed.

hcraigmiller, 2022-11-18:
From a technical standpoint, "out-of-date" is an adjective which precedes a noun and "out of date" may otherwise be used. However, grammatical sources (such as dictionaries) generally prefer the use of "out-of-date" in all contexts. I would prefer the latter as well.

SiboVG, 2022-12-01:
We'll need @neilweinstock for the final decision.

neilweinstock, 2022-12-01:
There are rules, apparently (or at least, "guidelines"): https://grammarhow.com/up-to-date-or-up-to-date/

JoePfeiffer, 2022-12-01:
> There are rules, apparently (or at least, "guidelines"): https://grammarhow.com/up-to-date-or-up-to-date/

Unfortunately when the phrase appears without context (as it does here), we don't have any clues as to whether it's "this data is out of date" or "this is out-of-date data".

neilweinstock, 2022-12-01:
Is there a place in OR where it is used to modify a noun?  I can't think of any.  If it says "Simulation 1 / up to date" then that would seem to fit the case described in the usage guidelines where it should *not* be hyphenated.

SiboVG, 2022-12-12:
Closing because the current terminology is fine.

closed 2022-12-12 by SiboVG