openrocket/openrocket #1833: Launch failure in macOS Ventura 13.0.1

SkierPiau, 2022-11-15:
OpenRocket simply doesn't launch, or launches but doesn't appear on screen.

[image]

[image]

Tried the solutions offered by this issue: https://github.com/openrocket/openrocket/issues/827 but doesn't work. Made sure access was enabled even though it is unsigned and the typical "unidentified developper" prompt from apple appears, after which I simply click on open, but the app doesn't launch.

JoePfeiffer, 2022-11-15:
Which version are you attempting to run?  If 15.03, download the current version from https://openrocket.info/downloads.html?vers=22.02.beta.05

SkierPiau, 2022-11-15:
Worked with the beta, thank you!

JoePfeiffer, 2022-11-15:
You're welcome!

closed 2022-11-15 by JoePfeiffer