openrocket/openrocket #1837: Export

benrussell11, 2022-11-16:
Use the design file from my "Chute" bug report.

Export Simulation 3
[image]
Stability margin in calibers and Mach Number are also selected
Drop down list - Stage to export
 - Selecting Booster exports Sustainer only
 - Selecting Sustainer exports Booster and Sustainer full flight.  This just should only export Sustainer
 - Suggest a third option - Full flight which would export the full flight
 - Or keep it simple and have just one option which would export the full flight

In the exported CSV file, some column headings have strange characters for columns that don't have any units. Although I didn't export all of the possible columns.

[image]

Beta 5 Windows 10
Ben

JoePfeiffer, 2022-11-16:
Not quite directly related to your issue, but are you using custom thrustcurves?  If so, would you mind posting them too, so we can recreate the situation you're seeing as accurately as possible?

JoePfeiffer, 2022-11-16:
I went ahead and ran your file with some standard motors to see if I could see what was happening --

What's going on with the different simulation branch exports is counterintuitive until you know what's going on, but we think it is the most effective for most uses (certainly for mine):

When you select export data for the sustainer, it exports the data for the sustainer's complete flight.  So it doesn't start at stage separation/sustainer ignition, it starts at takeoff.  That way we get the performance while the booster is burning, from booster burnout to sustainer ignition, and from sustainer ignition to landing in a single CSV.  For most users, that puts the data that's important in one place. Notice also that it's consistent with what the plot shows when you select "Sustainer".

When you select booster, it starts at stage separation and follows the booster for the rest of its flight.  We could duplicate the data from the sustainer branch when exporting the booster (and we do that in the plot), but we haven't seen much need for it.

The problem with exporting the full flight as a CSV is trying to distinguish between the sustainer and booster in the resulting file.  At some point I'd like to learn enough about Excel format to export into that, so we can have the whole flight as a single spreadsheet, with one simulation branch per sheet... but that won't happen for this release.

The odd characters you're seeing are a result of the flavor of Unicode -- we're exporting in UTF-8, and your spreadsheet program is trying to use something else.  So everything with a subscript, a superscript, a degree symbol... will be messed up.

I'll leave this open until I can look at it with your motors to make sure nothing odd is happening that I'm not anticipating.

benrussell11, 2022-11-16:
Hi Joe,

 

Attached is the eng file I’m using. SRAD motors for a 2-stage 10K flight.

 

Column headers – I’m using the default Windows character set. While I haven’t totally tested all columns, it looks like this problem is for columns which don’t have units. I would think the units for Mach number would be blank as there isn’t any that I know of.

 

For me at least, if I select an option from a drop down list, I expect the data to represent what I’ve chosen. If I select “Booster” then Booster data only. If I select Sustainer, then I expect to see only sustainer data. Maybe change it such that Booster gives you booster data,  Sustainer gives you sustainer data and something that indicates the complete flight. For a 3 stage flight, what would be the current options be?

 

While I have your attention. I need away to tell if the sustainer has tilted over by more than 20 degrees before sustainer ignition. Is there away to do this in OR?

 

Looking forward to having dinner with you at the cup.

 

Ben

benrussell11, 2022-11-17:
Hi Joe,

 

I’m looking at Angle of Attack plot plotting time vs angle of attach. The drop down shows All,  Sustainer and Booster Stage and does what I would expect.

 

It would be nice if we used different plotting colors for the booster and sustainer. Currently they are Blue.

 

Ben

JoePfeiffer, 2022-11-18:
Note that the plot for the Sustainer shows the same data as is exported -- all the data for the sustainer, not just the data after stage separation.  It does show more data for the Booster than is exported; for the Booster it only shows the data that's different from the Sustainer, ie after stage separation.

It's pretty common for people to plot many different variables at the same time.  Plotting the variables in different colors is more useful than plotting the stages in different colors (it would be really good to give users a choice of colors to plot in so they could make their own decisions on this and avoid things like plotting in yellow.  Maybe someday...).

I don't have excel set up (I'm generally not a Windows user); I don't know it's guessing so badly on the character encoding, but apparently you can go to an Options menu (in excel) and tell it what encoding to use.  UTF-8 should do the right thing.  There are probably other headings showing oddly, like degree symbols and superscripts.  The particular character showing problems in Mach is actually a zero-width space.

closed 2022-11-18 by JoePfeiffer