openrocket/openrocket #1851: "Edit Simulation" button text blanks when edit panel open

jimmiedave, 2022-11-24:
Mac OS Catalina, Intel Mac Book Pro 2019, beta 22b5

[image]

Not happening on Linux, Windows 7 (yes, really).

Steps to repro:

1. Open example ASMR.
2. Click "Flight Simulations" tab.
3. Click "Edit Simulation" button.

hcraigmiller, 2022-11-24:
Unable to reproduce.

Beta 5
Current Unstable

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

jimmiedave, 2022-11-24:
Yeah, not happening on Windows, Linux, as noted in first post.

PR #1856 (2022-11-28): [#1851] Reset button foreground on focus loss
This PR fixes #1851. Since this is a macOS-issue, @hcraigmiller can't test it so I'll just assume that my testing was adequate and it functions as expected.
closed 2022-11-28 by SiboVG