openrocket/openrocket #1853: Show stability range for all motor configurations

neilweinstock, 2022-11-28:
Presented for discussion for a future release.

Right now the only stability report you get is in the rocket figure for the selected motor config.  I think it would be more useful more of the time to show, somehow, the range of stability values for all configurations, including no motor.

There is no one obvious way to present the information.  One approach could be to show a list of stability values, with the current configuration highlighted in some way, when the user rolls over or clicks on something.  Other approaches might push towards a more significant reorganization of the UI.

In general, I think that much of the text that just hangs out in the rocket figure should be interactive.  Maybe clicking on the stability value is what brings up the larger summary.  There would need to be some sort of affordance to indicate that the text was clickable.

Multi-stage rockets present a more complex problem.  I'm not sure what the best way would be there.  I do think it would be nice to be able to see the stability of all stages at once, rather than needing to interactively disable stages to see the stability of the remainder of the stack.

Charles-McG, 2022-11-28:
I think there is a huge amount of information in a rocket-config-sim combination that could be brought out - both max/min/value@event within a combination and the same across combinations.  I'm usually more interested in variation across events, but we know that some folks use OR to compare configurations and some use the sim-settings extensively.

All the info is available in various graphs - which I suspect are under utilized.  I'd prefer more table (and exportable table) options, but I can see the value of having mini reports with more details available to drill into - making currently static screen elements interactive.

Personally, I think the Jony Ive design aesthetic has gone too far.  It may be that I'm just old - but I prefer interactive UI elements to be visually obvious and the active region I'm aiming the mouse at to at least give me hints that I'm in the zone.  I'm tired of guessing if on-screen text is just text, or if it will do something if I poke at it.

neilweinstock, 2022-11-28:
> I think there is a huge amount of information in a rocket-config-sim combination that could be brought out - both max/min/value@event within a combination and the same across combinations. I'm [...]

That would definitely be useful as well... information should be available wherever it is useful, not necessarily confined to one place in the UI.  Since the rocket figure is usually where stability is evaluated (I think), I would like to see more comprehensive info available there... but more detail in the sim results is good too.

However, the existing table format is not particularly conducive to having more data... lots and lots of columns just push the table off the screen and then demand a bunch of scrolling.  The sim results table could use a rework to allow it to show more information per sim more effectively (but let's save that discussion for a separate issue).

> All the info is available in various graphs - which I suspect are under utilized. I'd prefer more table (and exportable table) options, but I can see the value of having mini reports with more [...]

Agreed, 100%.

Charles-McG, 2022-11-28:
> However, the existing table format is not particularly conducive to having more data... lots and lots of columns just push the table off the screen and then demand a bunch of scrolling. The sim [...]

Like #1525 ?

neilweinstock, 2022-11-28:
Maybe, but that one doesn't really address the issue I'm talking about, which is that the way the information is displayed in the UI should be something other than a straight table like it is now.