openrocket/openrocket #1857: (beta 5): Opening file causes stack overflow error

neilweinstock, 2022-11-29:
Opening this file in beta 5 causes a crash.
[image]

[Saturn V Full Scale.ork.zip](https://github.com/openrocket/openrocket/files/10109313/Saturn.V.Full.Scale.ork.zip)

hcraigmiller, 2022-11-29:
What version was it created (will it open) with?

hcraigmiller, 2022-11-29:
You can create and save it, but you cannot reopen it.

[image]

It appears to be exceeding the maximum parameters of OpenRocket because of its size.

neilweinstock, 2022-11-29:
Interesting that it can be created and saved but not opened.  Allegedly it opens fine in 15.03.

Also: if I recall correctly, Sampo created a full-sized Saturn V model in OR some years ago just to show it could be done.  We should still be able to do that, even if that actual *usefulness* of such a model is limited.

PR #1861 (2022-12-01): Remove bounds on minimum and maximum zoom sizes
Trying to set a zoom smaller than the minimum size was provoking an infinite recursion.  It's not clear to me why we need min and max in the first place (the user can always use "Fit" to get back to something reasonable) so just get rid of them.

Fixes #1857
closed 2022-12-01 by SiboVG