openrocket/openrocket #1864: Add recovery option for complex parachute deployment

JoePfeiffer, 2022-12-01:
It's not uncommon for mid- and high-power rockets to have what we might call a "complex" parachute deployment:  a single parachute that opens in two distinct phases to implement dual deployment.  The Jolly Logic chute release is probably the most common example of this; another example is a reefed chute that is allowed to fully open later in the descent.

At present, this has to be modeled with two recovery devices: a streamer or small chute representing the chute in its reefed configuration, and a second chute with a size calculated so the total drag of  the two devices is the drag of the actual chute.

A better way to do this would be to have a "parachute" with two diameters: one giving the drag of the chute in its reefed state, and the second with the drag when fully open.  At the time of the first recovery event the chute is modeled using its closed diameter; at the second event the diameter is changed to that of the fully open chute.

There needs to be a clear warning that the user is pretty much on their own guessing the effective diameter of a chute held by a JL.  I wouldn't be at all surprised to find (which means I haven't looked) there is data out there on the performance of reefed chutes which could be used to inform that part of the implementation.

hcraigmiller, 2022-12-14:
Is Issue #1053 and this issue sufficiently duplicative so that Issue #1053 or this issue should be closed?

JoePfeiffer, 2022-12-14:
> Is Issue #1053 and this issue sufficiently duplicative so that Issue #1053 or this issue should be closed?

Yes.  This is a more general version, and supersedes #1053