openrocket/openrocket #1869: An unknown java error occurred

hamids23, 2022-12-04:
i,m using openrocket 22.02.beta.05 on ubuntu 22.04.1 LTS and i made a project, now i cant open it because of the error shown in below :
[image]
how can i fix it?

hcraigmiller, 2022-12-04:
Please post your design file. You can change the ".ork" extension to ".ork.txt" so that you can just drop it here. There is a known issue that causes error, which has been fixed, and the design file will help identify the issue whether known or new.

hamids23, 2022-12-04:
[rocket.ork.txt](https://github.com/openrocket/openrocket/files/10149134/rocket.ork.txt)

i'm openning openRocket using Terminal and when i checked the terminal and i see below lines printed several times on terminal, i think it's useful to figuring out what's going on

(java:5226): Gtk-WARNING **: 21:21:06.149: Theme parsing error: gtk.css:1422:23: 'font-feature-settings' is not a valid property name

(java:5226): Gtk-WARNING **: 21:21:06.158: Theme parsing error: gtk.css:3308:25: 'font-feature-settings' is not a valid property name

(java:5226): Gtk-WARNING **: 21:21:06.161: Theme parsing error: gtk.css:3770:23: 'font-feature-settings' is not a valid property name

hcraigmiller, 2022-12-04:
Unable to reproduce in either Beta 5 or current unstable.

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-12-04:
Can you verify if you can reproduce this error using [this JAR file](https://github.com/openrocket/openrocket/suites/9660500762/artifacts/462194225)? This issue sounds familiar to #1857, which was fixed by #1861.

hcraigmiller, 2022-12-04:
Unable to reproduce in Build 1250.

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-12-04:
Well I meant that @hamids23 should test it, because I can't replicate it either.

SiboVG, 2022-12-13:
@hamids23 can you please test whether you can replicate the issue using [this JAR](https://github.com/openrocket/openrocket/suites/9845700038/artifacts/474534591)?

wolsen, 2022-12-15:
> [rocket.ork.txt](https://github.com/openrocket/openrocket/files/10149134/rocket.ork.txt)
> 
> i'm openning openRocket using Terminal and when i checking the terminal, i can see below lines printed several times on terminal, i think it's useful to figuring out what's going on
> 
> (java:5226): Gtk-WARNING **: 21:21:06.149: Theme parsing error: gtk.css:1422:23: 'font-feature-settings' is not a valid property name
> 
> (java:5226): Gtk-WARNING **: 21:21:06.158: Theme parsing error: gtk.css:3308:25: 'font-feature-settings' is not a valid property name
> 
> (java:5226): Gtk-WARNING **: 21:21:06.161: Theme parsing error: gtk.css:3770:23: 'font-feature-settings' is not a valid property name

FWIW, these warnings are red-herrings and are not causing the failure that you are encountering. These are warnings that come from the Java runtime in the gtk environment (linux) and are not errors that are raised to the UI in the same manner.

I checked out at commit f529d3ee (the commit just prior to the fixes for #1861) but I was unable to recreate the issue using the provided ork. I think more information will need to be provided.

As a question to @hamids23, how did you install the OpenRocket version? Beta installer? Ubuntu App store (e.g. snap)? Jar file?

wolsen, 2022-12-22:
I'm closing this out for non-responsiveness and the inability to recreate this. Should more information be available, please feel free to add a comment and re-open this issue to be re-evaluated.

closed 2022-12-22 by wolsen