openrocket/openrocket #1890: [Bug] Zero-length warning should be zero-dimension warning

hcraigmiller, 2022-12-15:
When a phantom body tube is used, if either the diameter or length is set to zero then the "Zero-length bodies" warning is shown. The warning is actually shown if a body has either a zero length or a zero diameter. The "Zero" warning should reflect this and be changed to:

- Zero-length bodies may not result in accurate simulations,
- Zero-diameter bodies may not result in accurate simulations, or 
- Zero-length/diameter bodies may not result in accurate simulations.

In any event, the correct use of the hyphens should be implemented (for @SiboVG).

SiboVG, 2022-12-15:
> In any event, the correct use of the hyphens should be implemented (for @SiboVG).

😍

I would just go with "Zero-length/diameter bodies may not result in accurate simulations." to cover all scenarios.

hcraigmiller, 2022-12-17:
Fixed by way of PR #1891.

closed 2022-12-17 by hcraigmiller