openrocket/openrocket #1894: [Bug] Warning that inline pods "may not simulate accurately"

hcraigmiller, 2022-12-16:
An increasing number of users are positioning pods so that a component (I'll use a transition) can be placed inline as if the transition is the next airframe component after the body tube. @JoePfeiffer confirmed that from an aerodynamic standpoint, this technique may not be being simulated accurately.

[image]

With the increasing number of users implementing this technique, if there is a way to **_fairly easily_** determine whether a pod component is positioned in line with the main airframe, a warning should be given that "Pods with inline components may not simulate accurately." **_If not, this warning should be included in the Wiki._**

I am not suggesting aerodynamic correction. If I understood @JoePfeiffer, he does not believe that aerodynamic correction is feasible, even if possible, because of too many nuances in design variations and should not be undertaken.

JoePfeiffer, 2022-12-16:
I feel like this leads us down a rabbit hole of trying to anticipate every crazy thing a user can do; at some point, we just have to say "OR doesn't have a general geometry engine" and leave it at that.

hcraigmiller, 2022-12-16:
**No rabbit holes wanted here.** Only warning if and only if the pod position is relative to center of the parent component AND the radial distance of zero, and, most importantly, if and only if it can be accomplished with relative ease.

JoePfeiffer, 2022-12-20:
Just came across someone on FB recommending using pods to position things to get a simulation of a non-cylindrical body tube to simulate.  Never mind that I've got no idea how getting something to look right doing that would even work, it's clear you're right that it's a seductive mistake.  We do need the warning.

But the warning should be for anything involving in-line pods, not just when they're butting up against them.

hcraigmiller, 2022-12-20:
Agreed, only warn if the pod position is relative to center of the parent component AND if and only if the radial distance is zero.

neilweinstock, 2022-12-21:
> at some point, we just have to say "OR doesn't have a general geometry engine" and leave it at that.

This is incredibly important, but very hard to convey.  I have expressed this sentiment about 1000 times on TRF and (I think?) on the podcast.  I don't how to ensure that users really take it to heart, and I don't know how the program can detect geometries that will not simulate accurately, in the general case.

I do think it's useful to flag, when possible, invalid geometries that might be commonly use and readily detectable.  It's just going to be whack-a-mole with this stuff pretty much forever as far as I can tell.

SiboVG, 2023-01-21:
Was not fixed by #1979, #1979 was just linked to this issue by accident.

PR #1979 (2023-01-21): Warn for phantom tubes earlier on
This PR fixes #1894 and adds a warning when a pod set is in line with the rocket's center line. Question: should the same warning also be generated for boosters?

I also noticed that the zero-volume warning for phantom tubes was only added when fins were added to the phantom tube. I thought it would be better to generalize this error and check for zero-volume tube warnings even when there are no fins attached to the component.

I also added the phantom tube component name to the zero-volume warning.

https://user-images.githubusercontent.com/11031519/212781178 [... cut]
PR #1985 (2023-01-24): Check for gaps and overlaps in airframe
Since users are using pods with a radial offset of 0 to create sci fi rockets with exposed inner tubes and stuff, warn them that they aren't likely to simulate accurately.

Also, if a user sets up a rocket using a pod that doesn't have a gap in the airframe, we are able to simulate that so we should get rid of the warning in that case.

First screenshot:  a rocket with a single long body tube.
[image]

Second screenshot: rocket with the same geometry but two body tubes, using a PodSet to position the second one. Note CP and performance summary are roughl [... cut]
closed 2023-01-24 by JoePfeiffer