openrocket/openrocket #1895: [Bug] Component in component tree has different font size when overridden

SiboVG, 2022-12-17:
When a component has an overridden value, it's font size is suddenly increased.

[image]

hcraigmiller, 2022-12-17:
I'm seeing a slight elongation that is almost indiscernible, not really a font size change.

[image]

I tried every override combination with the same result.

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

SiboVG, 2022-12-17:
> I'm seeing a slight elongation that is almost indiscernible, not really a font size change.
>
> [image]
>
> I tried every override combination with the same result.
>
> [Windows 11 Pro; Version 22H2; OS [...]

Must be a macOS thing then 🤷‍♂️.

PR #1913 (2022-12-23): [#1895] Maintain tree cell font after override icon
This PR fixes #1895.

[image]