openrocket/openrocket #1898: [Bug] Simulation fails

hcraigmiller, 2022-12-17:
Open the attached design file and go to the Flight simulations tab. Add a simulation using the Estes E12-0 motor and close the pane without running the simulation. Then run the simulation from the Fight simulation tab.

[image]

[01.03.01.Motor Mount.24mm.Estes 95mm Engine Hook.ork.txt](https://github.com/openrocket/openrocket/files/10252661/01.03.01.Motor.Mount.24mm.Estes.95mm.Engine.Hook.ork.txt)

[Windows 11 Pro; Version 22H2; OS Build 22621.521; Windows Feature Experience Pack 1000.22634.1000.0]
[Java "11.0.15" 2022-04-19 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.15+8-LTS-149)]

JoePfeiffer, 2022-12-19:
I'm not sure this is even a bug...  I'll see about either putting a clamp on the simulation step and/or putting up an exception for a "rocket" that's got so little aerodynamic surfaces it can't be simulated.

hcraigmiller, 2022-12-22:
Fixed by way of PR #1914]

JoePfeiffer, 2022-12-22:
No, #1914 is just helping me find what's going on with this.  The simulation is still failing on an exception.

Trying to solve this one is an interesting exercise, but in spite of the labels I'm not sure it's really any kind of blocker.  Basically a totally ridiculous simulation fails.  My current plans on solving it are --

1. see if I can find a way to limit the sim step size to keep crazy aerodynamics from provoking a numerical instability.  This shouldn't be difficult, but I need to look at it seriously in the morning.
2. if not, see if I can characterize "crazy" in a way that can bring up a popup refusing to even try.  This one has a zero-length total external symmetric components so it would be easy to identify if the first idea is too hard; is there a more general characterization?
3. punt.  Anybody who tries to sim a naked motor mount really deserves what they get...

PR #1924 (2022-12-27): Throw exception at simulation startup if CP can't be calculated.
Exception says "Can't calculate rocket stability."

The basic problem with #1898 is it's impossible to calculate the flying motor mount "rocket"'s center of pressure except when it's flying backwards (at nearly 180 degrees).  This results in ever-faster gyrations until one parameter or another exceeds OR's limits.  I spent a little while seeing if I could adjust the simulation time step in response to the speed of the gyrations; the end of result of these experiments was I could calculate with ever-tinier timesteps which made the [... cut]
closed 2022-12-27 by SiboVG