openrocket/openrocket #1910: [Bug] Only one recovery device is calculated on descent when multiple boosters/pods each have a recovery device

hcraigmiller, 2022-12-20:
With the mass fixed (overridden), increasing and decreasing the number of boosters with parachutes does not change the descent rate accordingly. Only one of the booster parachutes is ever counted in the descent rate computation.

[image]

[Boosters.Recovery_Device_Test.ork.txt](https://github.com/openrocket/openrocket/files/10269637/Boosters.Recovery_Device_Test.ork.txt)

JoePfeiffer, 2023-01-05:
Trying to understand how you ran the multiple-chute tests -- did you duplicate the Booster set to get the second and third tests?

hcraigmiller, 2023-01-05:
I fixed the mass and ran the test with 1 booster, then 2 boosters, through 4 boosters. The descent rate was the same each time, even though there were more parachutes.

PR #1949 (2023-01-05): Consider number of parachutes in landing drag calculation
Fixes #1910

The .ork file associated with that bug report now shows descents at speeds of roughly 18.7, 13, and 10.8 fps for 1, 2, and 3 boosters in the booster set. 18/13 is approximately sqrt(2), and 18/10.8 is approximately sqrt(3) so this is expected.
closed 2023-01-05 by wolsen