openrocket/openrocket #1923: [Bug] Simulation extension error on load of examples

hcraigmiller, 2022-12-24:
The following examples are throwing simulation extension errors:

- Simulation Extensions
- Simulation extensions and scripting

[image]

SiboVG, 2022-12-30:
This is the result of PR #1873.

I would also suggest to merge the "Simulation extension" and "Simulation extension and scripting" example, since they're basically the same.

JoePfeiffer, 2022-12-30:
Rather than combine them, maybe we should separate them?  Having the Java extension do the roll control and air start in the extension example, and having the JavaScript script do both in the scripting example (currently a script is used for roll control, and an extension for air start)?

JoePfeiffer, 2023-01-05:
@hcraigmiller could you check to see if this is still happening for you?  It isn't for me.  I do still want to better separate the examples, as I suggested above, before we close this.

hcraigmiller, 2023-01-05:
Using the current unstable, both appear fully functional, not only when loaded, but when all of the simulations are run as well.

closed 2023-01-17 by SiboVG