openrocket/openrocket #1933: Suggestion

Ali1Almakhmari, 2022-12-29:
Is it possible to add a feature to export the rocket's coordinates into an excel or text file? That would be super helpful in future releases.

JoePfeiffer, 2022-12-29:
The capability to export rocket coordinates, and all other flight parameters, as a CSV file already exists.

After running a simulation, if you go to the Plot/Export dialog, you'll find an "Export data" tab.  This gives you the opportunity to export whatever you want; the rocket's coordinates are given by the three columns "Altitude", "Position East of launch", and "Position North of launch".

I typically just export everything, and then select what I want to examine in my spreadsheet program.

closed 2022-12-29 by JoePfeiffer

Ali1Almakhmari, 2022-12-30:
> ---- External Email: Use caution with attachments, links, or sharing data ----
>
> Closed #1933<https://github.com/openrocket/openrocket/issues/1933> as completed.

Hello Joe,

What I meant is not the rocket’s coordinates in space, what I meant is the rocket’s structural shape (x,y) coordinates. In other words, the contour of the rocket. Let me know if this is something that already exists in Open Rocket. If not, I hope your amazing team will work on implementing it in the future.

Best,
Ali

JoePfeiffer, 2022-12-30:
Ah, so you mean some sort of representation of the exterior of the rocket in (x, y, z) coordinates?  Trying to figure out how to do that in a meaningful way would be a challenge...

Something we are hoping to do in a future release will be .stl exports of components so they can be 3D printed.  That would at least be a step in the direction you're looking for.

(incidentally, my son got a CS degree at Purdue -- good school!)

Ali1Almakhmari, 2022-12-30:
hhhhh, small world. CS is a really competitive program. I would say CS ,ECE, and AAE (my major) were the toughest at Purdue.

Regarding the program, I don't work on Java (mainly Python and MATLAB), but I would love to take a look at code/functions related to shaping/drawing the rockets in Open Rocket. Do you know which ones are they in Github? I previously worked on a similar project to automate drawing rockets in Blender using Python scripts. Perhaps I can edit the Java scripts and see if I can extract the coordinates of the rocket's contour from them. Because theoretically, if you are drawing the rocket, then the X,Y coordinates should be easy to obtain (I can't say how easy it will be regarding the z coordinate cause it depends on how the drawing process was programmed).

JoePfeiffer, 2022-12-30:
Sure, it's all under swing/src/net/sf/openrocket/gui/scalefigure/ and /swing/src/net/sf/openrocket/gui/rocketfigure/