openrocket/openrocket #1948: Merge BasicTumbleStepper into BasicLandingStepper

JoePfeiffer, 2023-01-04:
The two simulation steppers used for parachute/streamer and tumble recovery are nearly identical.  It really ought to be possible to merge them.

Osiris34, 2023-01-04:
Hey can I help complete this task?

SiboVG, 2023-01-04:
> Hey can I help complete this task?

Sure, be our guest!

hcraigmiller, 2023-01-04:
See PR $1947, completed.

hcraigmiller, 2023-01-04:
But here are the current outstanding issues. Something there?
[Issues.Summary.2023-01-04.01.pdf](https://github.com/openrocket/openrocket/files/10347853/Issues.Summary.2023-01-04.01.pdf)
[Class.Summary.2023-01-04.01.pdf](https://github.com/openrocket/openrocket/files/10347855/Class.Summary.2023-01-04.01.pdf)

Osiris34, 2023-01-04:
Thank you, would you mind providing a little more information on what must be done to complete the task?

JoePfeiffer, 2023-01-04:
> Hey can I help complete this task?

We'd love it!  I'm assuming you're familiar with github, cloning repositories, pull requests, etc (if you're not, we'll be happy to fil you in).

Anyway, the issue I noticed has to do with simulation of a rocket in the landing phase of its flight.  The simulation itself is in the directory core/src/net/sf/openrocket/simulation/

It's built as a bunch of different simulation steppers, for different phases of flight.  There are two steppers that deal with the landing phase:  BasicLandingStepper.java, which handles simulations that have a parachute or streamer, and BasicTumbleStepper.java, which deals with simulations that have no recovery device.

These two steppers are nearly identical:  they both calculate the drag under the different conditions, and then perform an Euler integration to calculate a new position and velocity for the rocket.  It seems like it should be possible to combine them.  My first thought is maybe just have a single landing stepper which can use either drag calculation; another thought is to maybe pull the Euler integration out into something that can be called from both of them.

So, designing and implementing a solution for this would clean up a little bit of the code, and give you some familiarity with the code base.

Let me know if you have further questions!

JoePfeiffer, 2023-01-04:
> But here are the current outstanding issues. Something there? [Issues.Summary.2023-01-04.01.pdf](https://github.com/openrocket/openrocket/files/10347853/Issues.Summary.2023-01-04.01.pdf) [...]

I'm thinking he was asking specifically about this issue.

Osiris34, 2023-01-04:
Thank you. This is my first ever github contribution. Thank you for helping me.

JoePfeiffer, 2023-01-04:
OK, I see you've cloned the OR repository, so you're on your way.  Our typical workflow is:

1. create a new branch in your repository for the fix.  When I'm fixing an outstanding issue I normally name my branches something like fix-1234 so it ties it directly back to the problem, but the name doesn't really matter.
2. implement the fix in your local copy of the repository.
3. test your fix using the unittest build target
4. commit and push your fix
5. up on github, you'll see you have an option to create a Pull Request.  Create the PR, and we'll look it over.  just about everybody gets suggestions for improvements to the PR on their first several contributions.
6. One of the team members will merge the PR into the main 'unstable' branch.
And then it'll be time to tear off another issue...

Osiris34, 2023-01-07:
Hey, I have been working on this issue for a few days now and it appears that the two steppers are basically the same but their euler integrations are different and also am I calculating the drag force or drag acceleration or both? I may need some more help completing this task.

SiboVG, 2023-01-07:
> Hey, I have been working on this issue for a few days now and it appears that the two steppers are basically the same but their euler integrations are different and also am I calculating the drag force or drag acceleration or both? I may need some more help completing this task.

Hey, thanks for the effort so far! I'll ping @JoePfeiffer, he's more knowledgeable on this.

JoePfeiffer, 2023-01-07:
Ultimately, you want to compute a new velocity and position based on sum of the accelerations from drag, gravity, and coriolis force.  To get the drag acceleration, you need to calculate the drag force.

The Euler integrations are the same (unless I'm misreading the code, which is always possible).  One thing that does differ between the two is the BasicLandingStepper takes a look at the change in acceleration, and reduces the time step if it's too high -- otherwise we got into some pathological situations where the descent speed started oscillating.  I think you'll want to have that adjustment in the final resulting code.

GitteVandevenne, 2023-03-09:
Is there still help needed to complete this task?

SiboVG, 2023-03-09:
Hey Gitte (sounds like a Belgian or Dutch name? I'm Belgian myself). The issue is still open and it appears that @Osiris34 hasn't been able to work on this anymore, so I'd say you can give it a try!

JoePfeiffer, 2023-03-09:
Yes, please work on it!  I've thought about it a little bit since I made the original suggestion, so I think I can offer a bit more guidance than I did before.

The two classes are nearly identical and both use Euler integration, with the exception that their drag calculations are different.  So I think (feel free to disagree and suggest another alternative) that what's needed is an AbstractEulerStepper class; everything in this class is concrete except a calculateDrag method.  Now both BasicLandingStepper and BasicTumbleStepper extend AbstractEulerStepper, and define calculateDrag.

Please feel free to ask more questions about how to do this.

PR #2199 (2023-05-07): Pull common code from BasicLandingStepper and BasicTumbleStepper into new AbstractEulerStepper
Fixes #1948
closed 2023-05-07 by JoePfeiffer