openrocket/openrocket #1969: Uncaught Exception Loading Older ORK Files

tjkopena, 2023-01-14:
I have a variety of ORK files made within the past ~2 years that crash the parser in 22.02.beta.05, compiled from unstable 890d6de on Arch Linux. One of them is attached, along with the bug report from trying to load it.

[bug.txt](https://github.com/openrocket/openrocket/files/10417229/bug.txt)
[R1.ork.zip](https://github.com/openrocket/openrocket/files/10417232/R1.ork.zip)

SiboVG, 2023-01-14:
Can you run `ant clean` in the source dir and then build again?

SiboVG, 2023-01-14:
We've updated the `net.sf.openrocket.rocketcomponent.FlightConfigurationId` class name some time ago, and if you don't have a clean build, you can get that exception.

tjkopena, 2023-01-14:
Success! All my files are loading after cleaning & rebuilding. Thanks!

closed 2023-01-14 by tjkopena