openrocket/openrocket #1970: [Feature request] Change Inner Tube cluster tube separation from unitless to distance unit

SiboVG, 2023-01-14:
In the Cluster tab of an inner tube, the tube separation is set as a unitless value, where 1 is tubes touching, 2 is 2 tubes distance I guess etc. Wouldn't it make more sense to have this be an actual distance instead of the ambiguous mess that it is now.

jura0207, 2023-01-15:
Hi, can I work on this issue?

I managed already to change units to length and add an unit selector to the subpanel. Now I just have to configure it to be for a distance and not a scale.

JoePfeiffer, 2023-01-15:
Go for it!  Let us know if you have any questions on how to proceed.

SiboVG, 2023-01-15:
Thanks for your interest @jura0207! If I can give you a suggestion on your implementation, because it appears a simple task, but to do it properly, you'll need some background information.

You can see in `InnerTubeConfig.java` that the separation double model equals:
```Java
DoubleModel dm = new DoubleModel(component, "ClusterScale", 1, UnitGroup.UNITS_NONE, 0);
```
The `"ClusterScale"` key means that the double model will search for a `setClusterScale` method in the `InnerTube` class to set the inner tube's cluster scale, and the `getClusterScale` to get the cluster scale.

Internally, InnerTubes also have a `getClusterSeparation` method (declared in `core/src/net/sf/openrocket/rocketcomponent/Clusterable.java`). That's the method that we're after, since it declares cluster separation in units of distance.

So now your goal is to replace `ClusterScale` with `ClusterSeparation`. You'll need to create a `setClusterSeparation` method in `Clusterable.java` and implement `setClusterSeparation` in `InnerTube.java`. Now you need to get rid of the `clusterScale` variable inside `InnerTube.java` and replace it with a `clusterSeparation` variable. The `getClusterSeparation` and `setClusterSeparation` get/set the `clusterSeparation` variable. The `getClusterScale` method returns the cluster scale, transformed back from the `clusterSeparation` variable (the reverse operation of what's currently implemented in `getClusterSeparation`). The `setClusterScale` sets the `clusterSeparation` variable using the same back-transformation.

The reason why you still need `getClusterSeparation` and `setClusterSeparation` is for backward compatibility reasons with OpenRocket 15.03 (more on this later).

Now there are some cautionary areas that need changes:
1. Document saving
2. Document importing
3. Component scaling
4. RockSim saving
5. Other

**1. Document saving**
The .ork design file saves the cluster separation using the cluster scale. We want to change that to cluster separation, but we also want to maintain backward compatibility between OpenRocket 15.03. By this I mean that if you save a design file in OpenRocket 22.02, OpenRocket should (as much as possible) still be able to read that 22.02 design file. If you were to save the cluster separation, and discard the cluster scale, then OpenRocket 15.03 will not be able to read it.

To cope for this, you will need to save the cluster separation twice, but in two **different formats**: once as `clusterscale`, so OpenRocket 15.03 can read it, and one as `clusterseparation`, which is the format that we eventually want to keep.

So in `core/src/net/sf/openrocket/file/openrocket/savers/InnerTubeSaver.java` you will need to implement this:
[image]

**2. Document importing**
Analogous to the document saving, OpenRocket 22.02 should be able to read old design files that still have the `clusterscale` parameter. So in `core/src/net/sf/openrocket/file/openrocket/importt/DocumentConfig.java` you need something like this:
[image]

**3. Component scaling**
You can scale an inner tube by selecting it in the component tree, and then selecting in the application ribbon: "Edit > Scale". The `clusterScale` parameter does not need to be scaled, because it is independent of the component size (it's a unitless parameter). However, now you switched to the `componentSeparation` parameter, which does need to be scaled. So, in `swing/src/net/sf/openrocket/gui/dialogs/ScaleDialog.java` you need to add: 
[image]

**4. RockSim saving**
OpenRocket can export .ork files to a RockSim file. I haven't looked at the code of how OR does this for inner tubes, but the change to `clusterseparation` may have broken something. If you don't have RockSim to test something out, don't worry. Either give a quick look at the code to find obvious mistakes, or just remind us to test the RockSim saving when you create your Pull Request on this issue.

**5. Other**
I suggest you do a general file search in the source code for "clusterscale" to find other code areas that need some changes. E.g. I did a "clusterscale" search and found the following code:
```Java
public static InnerTube makeIndividualClusterComponent(Coordinate coord, String splitName, RocketComponent theInnerTube) {
  InnerTube copy = (InnerTube) theInnerTube.copy();
  copy.clearConfigListeners();
  copy.setClusterConfiguration(ClusterConfiguration.SINGLE);
  copy.setClusterRotation(0.0);
  copy.setClusterScale(1.0);     <--- bingo!
  copy.setRadialShift(coord.y, coord.z);
  copy.setName(splitName);
  return copy;
}
```
Here you see the code `copy.setClusterScale(1.0);`. I would replace this with `copy.setClusterSeparation( <inner tube diameter> );`. In fact I would try to replace as much as the `clusterscale` occurrences as possible. Only for backward compatibility reasons, you can keep `clusterscale`. When we don't want backward compatibility anymore, we're gonna delete all the code references to `clusterscale` anyways.

---

I'm sorry @jura0207  for the very long explanation, and am equally sorry if I gave away too many answers that you wanted to discover yourself. Just want to make sure that you don't get overwhelmed with weird bugs, and that it's done properly. Also, if some of my demo code is wrong, I apologize 🙃.

jura0207, 2023-01-15:
Thank you @SiboVG for a detailed issue analysis! It's very helpful. I managed to work something out by myself. I replaced scaling with distance and it worked, but I got funny readings while opening files, because .ork file still had scaling as input.

Now that you gave me a notice about the backwards compatibility and some tips I have some idea how to do it.

With distance now working, I can add on a function where a file is loaded an _if case_ to see if the file has clusterScaling and clusterSeparation, then with getOuterDimeter we can calculate and add clusterSeparation if it's missing or vice versa for the backwards compatibility. This way there is no need for clusterScaling.
Also, on saving, this way we can calculate and set clusterScaling too.

This way when the support for OR 15.03 stops, there is minimum changes to be made. There would be no parallel functions, getters and setters for scaling and distance, but only for distance. And clusterScale would be converted. I can try it this way and I'll report my results back.

Thank you for giving a notice to backward compatibility, I forgot about that.

For testing RockSim simulation, I saw that there are some default rocket models, could I test it with those?

jura0207, 2023-01-15:
I got progress. It turned out I have to use setClusterScale and in it I'm also calculating clusterSeparation out of the scale:
[image]

After reading clusterScale, I'm reading clusterSeparation:
[image]

It saves both clusterScale and clusterSeparation:
[image]

SiboVG, 2023-01-15:
> I got progress. It turned out I have to use setClusterScale and in it I'm also calculating clusterSeparation out of the scale: [image]
> 
> After reading clusterScale, I'm reading clusterSeparation: [image]
> 
> It saves both clusterScale and clusterSeparation: [image]

It's better for `setClusterScale` to be:
```Java
public void setClusterScale(double scale) {
    double separation = scale * getOuterRadius() * 2;
    setClusterSeparation(separation);
}
```

then you can keep all the logic inside the `setClusterSeparation` method.

SiboVG, 2023-01-15:
And in the parameter saving, don't type out the formula for cluster scale, but call the method `tube.getClusterScale()` and then have that method be
```Java
public double getClusterScale() {
    return getClusterSeparation() / (getOuterRadius() * 2);
}
```

You want to maintain all the logic inside `InnerTube.java`, that's just good practice.

jura0207, 2023-01-15:
Yes, thank you! I believe that's the better way. I've changed it, tested the software and everything works as expected. I've tested loading files with only clusterScale and both clusterScale and clusterSeparation, I've tested scaling, resetting, creating new cluster and spliting clusters. I haven't tested changes in RockSim.

Are there some documentation changes that I should change?

SiboVG, 2023-01-15:
> Yes, thank you! I believe that's the better way. I've changed it, tested the software and everything works as expected. I've tested loading files with only clusterScale and both clusterScale and [...]

Great! If you're ready with everything, you can push your PR.

> Are there some documentation changes that I should change?

You should delete this:
[image]

plus all the `InnerTubeCfg.lbl.ttip.TubeSep` translation keys inside the messages-files in `core/resources/l10n`
[image]

jura0207, 2023-01-15:
Language changes implemented and pull request created

PR #2269 (2023-07-27): [#1970] Add absolute inner tube separation option
This PR fixes #1970 and is a continuation of the work done by @jura0207 in #1974. You can now choose to set the inner tube cluster separation in relative or absolute units.

Demo:

https://github.com/openrocket/openrocket/assets/11031519/7e236462-91ba-4af5-89bb-3c75fe1bb79d
closed 2023-07-27 by SiboVG