openrocket/openrocket #1972: Update example rockets to include simulation data

SiboVG, 2023-01-15:
The current example rockets aren't all saved with simulation data included. E.g. if you open "A simple model rocket" and go to the simulation's tab, you can see that the simulations are marked as out-of-date.

hcraigmiller, 2023-01-15:
This issue may be interrelated with Issue #1973. For example, using the _Pods--airframes and winglets_ example:

[image]

The three example designs that seem to be affected by this are: (1) Dual parachute deployment, (2) Pods--airframes and winglets, and (3) Simulation extensions and scripting. The other example designs do not exhibit this behavior.

SiboVG, 2023-01-16:
> This issue may be interrelated with Issue #1973. For example, using the _Pods--airframes and winglets_ example:
>
> [image]
>
> The three example designs that seem to be affected by this are: (1) Dual [...]

Maybe a dumb remark, but you did enable "Save all simulated data" right?
[image]

hcraigmiller, 2023-01-16:
Good catch. . . ignore my oversight. . . it's been a long few days.

SiboVG, 2023-01-16:
> Good catch. . . ignore my oversight. . . it's been a long few days.

No worries, I'm just happy that there's no new bug 🙂.

closed 2023-01-17 by neilweinstock