openrocket/openrocket #1986: [Feature request] Add display drag value for recovery devices

hcraigmiller, 2023-01-18:
To compute "snatch force" when a recovery device opens, the recovery device drag force must either be known or calculated. Where OpenRocket is already calculating that value, _displaying_ the drag value used by OpenRocket would simplify the process of calculating the snatch force.

Suggested placement:

[image]

JoePfeiffer, 2023-01-19:
We are actually calculating that (a worst-case assuming the chute opens instantaneously) as part of the landing stepper.

It really has  to go in the simulation results somewhere, since it varies with the deployment velocity.

hcraigmiller, 2023-01-19:
Speaking for myself, I would like to be able to see the recovery device drag, being able to see how it's drag is affected by changes in the recovery device configuration. The amount of drag has at times altered my recovery device selection. 

That said, it would be fantastic to see snatch force on the simulation results, especially where it's a currently calculated value that just needs to be added to the table. (See, Issue #1987.)

JoePfeiffer, 2023-01-19:
The drag force is already available, both for plotting and exporting.  For plotting, just add an extra Y value and select the drag force.  If you do this with the dual-deploy example you'll see that when the main chute deploys there's a momentary force of over 500 Nt, which gets reduced to under 20 Nt in about a second.  And, of course, once the rocket is descending at near-constant speed the drag force is equal to the weight of the rocket.

hcraigmiller, 2023-01-19:
After thoroughly reviewing the comments of @JoePfeiffer, and following his approach, I am satisfied that the information needed to calculate snatch force is reasonably available without adding the recovery device drag value to the component configuration pane.

Thank you @JoePfeiffer for your comments, assistance, and patience.

closed 2023-01-19 by hcraigmiller