openrocket/openrocket #1988: Phantom body tubes are reverting to default diameter

neilweinstock, 2023-01-19:
This previously happened to my pods design example rocket... now it has happened to one of my designs that I haven't opened up in a long time.  When the phantom tubes suddenly take on a 1.9" diameter, the rocket looks like an exploded view:
[image]

I don't know if the file itself got corrupted or if OR simply opened it incorrectly... but since this has now happened twice, it seems to be an issue.  Here's the file for the above image.  I will check my other files and see if this occurs anywhere else.
[xb70.ork.zip](https://github.com/openrocket/openrocket/files/10452681/xb70.ork.zip)

hcraigmiller, 2023-01-19:
Duplicate of Issue #1975.

neilweinstock, 2023-01-19:
I haven't seen it in any other files... although for this one, it is repeatable.

hcraigmiller, 2023-01-19:
I have reviewed all of the merges since 2022-12-25 and it appears that the error in the Pod example has existed since it was first merged on 2022-12-28. And, when I changed the file names, rocket design names, and menu names to match, I failed to notice that the error was present, so it persisted. 

Once an affected file has been opened, fixed, and re-saved, I have been unable to reproduce the error. This has been true for the pod example and for the file attached above.

Moreover, what has occurred is not affecting all pods with phantom tubes, only body tubes attached to pods referenced to the surface of the parent component seem to be affected.

I hope this helps.

neilweinstock, 2023-01-19:
That's weird about the pod example... I know it was correct when I saved it.

SiboVG, 2023-01-19:
I've never been able to replicate issue #1988 or #1975. It appears we have a Heisenbug. @neilweinstock can you reproduce the error consistently, say 1 in every X times you open it? And are you using the latest unstable, or 22.02.beta.05?

neilweinstock, 2023-01-19:
It happened for me on both beta5 and a more recent unstable build. I will test some more. 

Are you saying the attached design opens correctly for you?

hcraigmiller, 2023-01-19:
No, I’m saying that if you open it with the current unstable, fix it, save it, and reopen it with the current unstable, then the error no longer exists and cannot be replicated.

SiboVG, 2023-01-19:
> Are you saying the attached design opens correctly for you?

Yes, I have no issues whatsoever.

SiboVG, 2023-01-19:
If I can't replicate the issue it looks like you'll have to fix the Java code yourself @neilweinstock 😉.

SiboVG, 2023-01-19:
Somehow you have set the radius of the body tube to "auto". In the OR code, if the `getOuterRadius()` code if `autoRadius` is `true` and there is no previous or next component to the body tube (which is the case), OR returns the default radius.
[image]
[image]

SiboVG, 2023-01-19:
I actually think the "Auto radius" should not be possible for the canard body tube. The auto radius function should only work within the parent pod set, it should NOT get its auto values from parts outside 'Canard mount'.

SiboVG, 2023-01-19:
@neilweinstock can you verify whether the "Auto" checkbox is still checked in the body tube of the canard and wings? I can only reproduce this issue when those checkboxes are unchecked.

hcraigmiller, 2023-01-19:
> I actually think the "Auto radius" should not be possible for the canard body tube. The auto radius function should only work within the parent pod set, it should NOT get its auto values from parts outside 'Canard mount'.

When I open the file in the current unstable, even though the code has radius set to auto for the outer diameter, the automatic box is greyed out on the configuration menu.

SiboVG, 2023-01-19:
> > I actually think the "Auto radius" should not be possible for the canard body tube. The auto radius function should only work within the parent pod set, it should NOT get its auto values from parts outside 'Canard mount'.
> 
> When I open the file in the current unstable, even though the code has radius set to auto for the outer diameter, the automatic box is greyed out on the configuration menu even though it is showing the default value.

Ow yeah you're right, in 22.02.beta.05 it isn't for some reason. I can replicate this issue in the current unstable.

In that case: is this still an issue? It appears to be an issue for designs saved in OR 22.02.beta.05, where the body tubes were allowed to have the auto radius, which it is not the case in the latest unstable. I have not tested this thoroughly, this is just my first impression.

hcraigmiller, 2023-01-19:
It appears to be a Beta 5 bug (option change) that has been fixed in the current unstable.

There's no way to go back in time and fix the designs that were saved with the auto setting; the only option would be to test to see if the automatic box/setting is "available/allowed" when a file with an auto radius is opened and, if not, set the value to zero.  However, I'm not sure that this approach won't result in its own set of problems down the line. It might be best left as is, with a Wiki note.

SiboVG, 2023-01-19:
> However, I'm not sure that this approach won't result in its own set of problems down the line. It might be best left as is, with a Wiki note.

My thoughts exactly...

hcraigmiller, 2023-01-20:
Wiki commented: [Why does my phantom body tube outer diameter change when I open a save design?](http://wiki.openrocket.info/FAQ#Why_does_my_phantom_body_tube_outer_diameter_change_when_I_open_a_saved_design.3F)

neilweinstock, 2023-01-20:
I sure don't think that I would have ever created a phantom body tube by using the "automatic" diameter option... but I can't rule it out.  If it came up as "automatic" by default, and that put the diameter at 0, then maybe I just would have left it like that, not realizing what I was doing.

If that in fact is the reason, then I would agree that it's a non-issue and we can close this.  I'll close it now, and we can see if it ever pops up again in a different circumstance than what we're seeing now.

closed 2023-01-20 by neilweinstock

SiboVG, 2023-01-20:
> I sure don't think that I would have ever created a phantom body tube by using the "automatic" diameter option... but I can't rule it out. If it came up as "automatic" by default, and that put the diameter at 0, then maybe I just would have left it like that, not realizing what I was doing.

https://user-images.githubusercontent.com/11031519/213589456-15fb3ecb-19e6-4ff6-a986-53062d1cbb74.mov

Yeah it checks it automatically. So it's good that you closed this issue.