openrocket/openrocket #1991: [Bug] Default graphics editor is no longer recognized

hcraigmiller, 2023-01-21:
To edit graphics, the user must specify the graphics editor, selecting the "default" editor triggers an error. Huge problem.

[image]

SiboVG, 2023-01-21:
I can't replicate this issue on macOS. Is this error reported on your machine @hcraigmiller? And did it work in previous versions?

hcraigmiller, 2023-01-21:
Yes, and yes.

JoePfeiffer, 2023-01-21:
Hmmm..... trying it on Linux I get a dialog labelled "Edit decal" with a button saying "Select Graphics Editor Program", which doesn't appear to do anything (I click the button and get no visual response at all).

I see in your screen shot that you're getting a prompt saying the editor can be set in Preferences.  I'm not finding that option on my system.

Checking... for me, 15.03 does exactly the same thing.

hcraigmiller, 2023-01-21:
Hmmm. . . too. It appears that it last worked right before I upgraded to Windows 10. Since that upgrade, not even 15.03 will call up the graphics editor from the default setting.

SiboVG, 2023-01-21:
Is this a Windows 11 issue? There was no default image editor selected on my machine. If I right-clicked on the file > "Open with" > "Choose another application", I could see that Paint was my default editor:
[image]

However, if I tried to double-click the file in Explorer, Windows asked me which program I wanted to open it in. If I again did "Open with" > "Choose another application" and then clicked the "Always" button (while Paint was still selected), Windows now correctly associated all .png files with Paint. If I then tried the "Edit" texture button in OR, it functioned normally.

In other words: I think this is not an OR bug. You just need to select a default image editor application in Windows. There's not really a way for OpenRocket to fix this I think.

hcraigmiller, 2023-01-21:
Okay, I can replicate what you did, and it functions as well, but only if MSPaint is set as the default editor. If I associate a .png file with Paint 3d (a Windows application that is not an executable program), or Adobe (which is an executable program), it does not work. Apparently, OR only allows MSPaint to be designated as the default graphics editor.

hcraigmiller, 2023-01-21:
Does this have anything to do with this issue?

[image]

SiboVG, 2023-01-21:
> Okay, I can replicate what you did, and it functions as well, but only if MSPaint is set as the default editor. If I associate a .png file with Paint 3d (a Windows application that is not an [...]

Do you mean Adobe Acrobat? I don't think that is recognized as an image editor. 

It's strange that Paint 3D doesn't work however.

hcraigmiller, 2023-01-21:
> Do you mean Adobe Acrobat? I don't think that is recognized as an image editor.

I was referring to your Adobe Photoshop (I saw it on your screenshot), but Adobe Acrobat Pro is an image editor. For grins, I set that as my .png default editor but it did not work in OR as the default editor. Only MS Paints works as a default editor.

SiboVG, 2023-01-21:
> Does this have anything to do with this issue?

Hm, that looks like it wouldn't be a good cross-platform solution, I'm hoping to get the Desktop-solution working.

SiboVG, 2023-01-21:
> I was referring to your Adobe Photoshop (I saw it on your screenshot), but Adobe Acrobat Pro is an image editor. For grins, I set that as my .png default editor but it did not work in OR as the default editor. Only MS Paints works as a default editor.

Ah yeah well I don't have Photoshop installed on my Windows machine. I'm running Windows as a virtual machine on my Mac. The Photoshop application listed is that installed on my macOS system, it won't launch inside Windows. So I can't test whether Photoshop opens on Windows.

JoePfeiffer, 2023-01-21:
Ah, OK...  over on Linux, if the "Use default" checkbox is checked, I am able to click the "Edit" button with the results I described before.  If I uncheck the box, the Edit button is no longer live, but the "Texture:" combobox to its left becomes live.  At this point I can select a file to use as a texture (I don't have any ability to filter the files -- I only get "all files").  This brings up the "Edit decal" dialog; the "Select Graphics Editor Program" button remains unresponsive.  Now if I click "OK" I get the error message @hcraigmiller describes in the original bug report.

SiboVG, 2023-01-22:
> Insofar as Paint 3D is concerned, it has no executable file or known executable name; if it has one, it is apparently well hidden by Microsoft.

I guess people who want to use Paint 3D need to set up a custom command:
[image]

SiboVG, 2023-01-22:
@JoePfeiffer regarding your issue on Linux, I think either Desktop is not supported, or the edit action is not supported:
[image]

Just checked on my Linux build and I don't have the Graphics editor option in the preference window.
On macOS:
[image]
On Linux:
[image]

And clicking the "Edit" button shows some really weird widgets.
On macOS:
[image]
On Linux:
[image]

Based on this conversion I see two things that need to be changed:
1. Don't display the edit button in the appearance tab if the condition `Desktop.isDesktopSupported() && Desktop.getDesktop().isSupported(Desktop.Action.EDIT)` is not met (you can find this condition in `swing/src/net/sf/openrocket/gui/dialogs/preferences/GraphicsPreferencesPanel.java` line 64)
2. Add an image file filter in the "Open texture file" popup window.

I'm still not entirely sure what to do with the Windows issue. I want there to be a better warning to users, something like "Make sure you have a default program selected for '.png' files" for example. But I'm not sure.

neilweinstock, 2023-02-03:
Pushing this to future release.