openrocket/openrocket #1997: [Feature request] Allow multi-sim exporting

SiboVG, 2023-01-25:
While #1485 is about plotting multiple simulations, this is about the ability to export multiple simulations. The implementation between the two is different enough for this to be a separate issue, instead of adding it to #1485. Request comes from [this TRF post](https://www.rocketryforum.com/threads/announcement-openrocket-version-22-02-public-beta-5-is-now-available.171295/post-2380036).

neilweinstock, 2023-01-25:
To clarify: request is for an export of the results table, not the full dataset.  This is really a 3rd kind of export, and likely requires a different UI.

The function could either be a simple "Export all results", or it could be tied to the list of sims that are selected.  The former is probably easier for the user most of the time.  Could also work both ways: if no sims selected, export all.  If any sims selected, just export those.

neilweinstock, 2023-01-25:
Best output format is uncertain.  Refer to discussion on TRF starting with post Sibo linked above.