openrocket/openrocket #1999: Too many warnings displayed in rocket figure

neilweinstock, 2023-01-27:
I believe this image speaks for itself:
[image]

We should do something about this before release.  Since there's obviously not time to do any sort of major rethinking of the warning system (we can save that for later), is there perhaps a way to put the warnings in a box and make it expand/collapse, so you can get it all out of the way?  I'll try to come up with a detailed proposal.

neilweinstock, 2023-01-27:
Here's a description of a proposal.  I'll try to mock it up ASAP.

1) If there are no warnings, you display nothing, just like today.
2) If there is one warning, you display it, just like today.
3) If there is more than one warning, then you display a message like "There are _n_ warnings.  Show".  A clickable "expand" button of some sort is provided.  
4) Clicking on the expand button shows the full list of warnings as in the above image.  At that point, next to the "Warning: text (which should say "Warnings:" BTW) you put a "collapse" (or "hide") button.

All this would happen in the lower right, where the warnings are now, so no new space needs to be allocated in any toolbar or anything.

It could be that you display up to 2 or 3 warnings before offering the expand/collapse mechanism.

SiboVG, 2023-01-27:
I don't think it's possible to add fold-open buttons in the rocket view... As far as I understand, you can only really render text to the frame.

neilweinstock, 2023-01-27:
OK, in that case we'll have to put a control somewhere else.  What about on the right side of the status bar at the bottom, which is blank right now?

In that case it would say "Show _n_ warnings", or "Hide warnings".  or nothing if there are no warnings.  Warnings would be... I guess shown by default, since that's what people are used to?

SiboVG, 2023-01-27:
Yes, we can put something in the bottom right corner. A checkbox to hide all warnings could work. Maybe also a button to show all warnings, which would then pop up a dialog with a table with all the warnings?

neilweinstock, 2023-01-27:
I'm concerned that a popup dialog is too much change for the last minute.  We can consider such things in the future.  For now, we just need to able to hide all that stuff when needed.

I'm not sure if a checkbox is ideal, or what the text should say, but it would be a fine way to get started and then we can refine (quickly!) once we can stare at it in use.

JoePfeiffer, 2023-01-27:
Don't forget the real problem is that it's a very problematic design with *lots* of issues so it shows lots of warnings...

A show/hide warnings checkbox would probably be OK -- but any time warnings are hidden there needs to be a meta-warning stating the number of warnings that are being hidden.

hcraigmiller, 2023-01-27:
This is a thought. . . Should be easy to implement too.

[image]

neilweinstock, 2023-01-27:
> Don't forget the real problem is that it's a very problematic design with _lots_ of issues so it shows lots of warnings...

Absolutely, but it should still be possible to work with it without the warnings making it unusable.
 
> A show/hide warnings checkbox would probably be OK -- but any time warnings are hidden there needs to be a meta-warning stating the number of warnings that are being hidden.

Also agree.

> This is a thought. . . Should be easy to implement too.

Two downsides: 
1) No room to show any sort of visual indication of number of warnings, or anything like that.
2) Warnings are in lower right, not obvious to look top left (or top middle) for control.

HOWEVER, maybe this could solve it.  Put that checkbox there.  But when warnings are hidden, put a message in lower right that says "_n_ warnings hidden", right where the warnings are now.

hcraigmiller, 2023-01-27:
> HOWEVER, maybe this could solve it. Put that checkbox there. But when warnings are hidden, put a message in lower right that says "_n_ warnings hidden", right where the warnings are now.

That would definitely work.

neilweinstock, 2023-01-27:
Ultimately I think this is not exactly perfect, but I think it would hold us over until we can do something more elegant.
[image]

hcraigmiller, 2023-01-27:
You forgot to uncheck the "Show" box. . .

hcraigmiller, 2023-01-27:
The "show" buttons should probably be together.

neilweinstock, 2023-01-27:
> The "show" buttons should probably be together.

You mean stacked?  It's a little crowded there, hard to squeeze it in like that. 

UNLESS... you get rid of the "Show" label, stack the two buttons, and include the word "Show" in each of them.  I don't know if this is better (it's less consistent, but it doesn't clash) but it's an alternative:
[image]

hcraigmiller, 2023-01-27:
No, not stacked, I like the first way.

SiboVG, 2023-01-28:
I would still like it to be in the bottom-right corner actually, that way it's also close to the actual warning texts (which is considered good UI design (having the action and consequence be close together).
[image]

hcraigmiller, 2023-01-28:
Works for me.

neilweinstock, 2023-01-28:
Make it so!

closed 2023-02-01 by neilweinstock