openrocket/openrocket #2005: [Bug] Intermittent linking of pod overrides

hcraigmiller, 2023-01-30:
While working on the base-drag hack example, I experience two separate pods being linked such that the mass, Cg, and Cd of pod 2 was being overridden by pod 1 (causing the pod 2 overrides to be greyed out) and unchecking the overrides on pod 1 did not restore control of the pod 2 overrides to pod 2. But saving and reopening the design fixed the problem (restored control of the pod 2 overrides to pod 2).

Happened only once, just FYI.

JoePfeiffer, 2023-01-31:
See if you can find a way to reproduce it.

hcraigmiller, 2023-01-31:
I spent an hour and a half trying to get it to do it again. I don't whether to be thankful or not that I couldn't reproduce it at this point.

SiboVG, 2023-01-31:
Heisenbugs, the worst kind of bugs...