openrocket/openrocket #2009: [Bug] Exception when clicking "Cancel" in rocket config on document load

SiboVG, 2023-02-01:
When loading a new design where the rocket config dialog pops open, if you click "Cancel", an exception is thrown.

https://user-images.githubusercontent.com/11031519/216088642-bdc0ac84-79fb-44cd-bd3b-283374d7f507.mov

hcraigmiller, 2023-02-01:
Unable to reproduce issue.

Build 
Java Version "11.0.17" 2022-10-18 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.17+10-LTS-269)
Microsoft Windows 11 Pro; Version 10.0.22621 Build 22621.1105; Windows Feature Experience Pack 1000.22638.1000.0

SiboVG, 2023-02-01:
> Unable to reproduce issue.
> 
> Build: Current Unstable Java Version "11.0.17" 2022-10-18 LTS; Java(TM) SE Runtime Environment 18.9 (build 11.0.17+10-LTS-269) Microsoft Windows 11 Pro; Version 10.0.22621 Build 22621.1105; Windows Feature Experience Pack 1000.22638.1000.0

Did you follow these steps:

1. Go to File > Open Example > A simple model rocket
2. The design opens with the rocket configuration window popping up
3. Click "Cancel"

hcraigmiller, 2023-02-01:
Correct, does not reproduce.

SiboVG, 2023-02-01:
> Correct, does not reproduce.

Ah, right, I know why. I set my cancel confirmation dialog to "Do not show again". You probably don't.

hcraigmiller, 2023-02-01:
Ahh, yes, there it is.

PR #2011 (2023-02-02): [#2009] When canceling a component change, check for an unmodified component earlier on
This PR fixes #2009. I also added a checkbox in the preferences to enable/disable the "ignore discard confirmation dialog" behavior:
[image]