openrocket/openrocket #2012: PDF export is not clear

Filipposcmd, 2023-02-01:
The exported PDF is sometimes not clear. I made a design for a small rocket, and  then i saw the exported PDF. It wasn't too clear. I attached it.
[rocket1.pdf](https://github.com/openrocket/openrocket/files/10561757/rocket1.pdf)

hcraigmiller, 2023-02-01:
Please post the .ork file too.

Filipposcmd, 2023-02-02:
Especially for small rockets

Filipposcmd, 2023-02-02:
> Please post the .ork file too.

WeTransfer link: https://we.tl/t-4PGk9aBgGe
(.ork file is not supported by github)

Filipposcmd, 2023-02-02:
Also i would like to join the translator team, i know both Greek and English (and i don't require payment)

hcraigmiller, 2023-02-02:
Just change the design file extension from ".ork" to ".txt" (or add .txt after .ork) and post the design file here.

Filipposcmd, 2023-02-02:
[Uploading rocket1.txt…]()

Filipposcmd, 2023-02-02:
[rocket1.txt](https://github.com/openrocket/openrocket/files/10573506/rocket1.txt)

hcraigmiller, 2023-02-03:
[image]

If not, provide a screenshot of what is unclear to you.

Filipposcmd, 2023-02-03:
[image]
[image]

If you ask me, i do not found this information clear.

hcraigmiller, 2023-02-03:
To the extent that you are referring to the overlapping text, this is a known issue, Issue #41. Otherwise, this is the standard fin template layout used by most model rocket manufacturers. How would you suggest making the information clearer?

neilweinstock, 2023-02-03:
> > Please post the .ork file too.
> 
> WeTransfer link: https://we.tl/t-4PGk9aBgGe (.ork file is not supported by github, and the model is unfinished)

In the future, the best way to post ORK files here (IMHO) is to zip them first.

neilweinstock, 2023-02-03:
I think in general there is plenty of room for improvement in the PDF output.

I'm trying to understand, though: for this particular issue, is it just that the function of the fin marking guide is unclear?

The template page is also missing some useful annotations.

SiboVG, 2023-02-03:
> In the future, the best way to post ORK files here (IMHO) is to zip them first.

Amen!

Filipposcmd, 2023-02-03:
How can i join the translator team i know Greek and English *and i do not require any kind of payment*

SiboVG, 2023-02-03:
> How can i join the translator team i know Greek and English _and i do not require any kind of payment_

We replied to your message on Slack.

Filipposcmd, 2023-02-03:
> > How can i join the translator team i know Greek and English _and i do not require any kind of payment_
> 
> We replied to your message on Slack.

ok, thanks

hcraigmiller, 2023-02-03:
See also, Issue #41.