openrocket/openrocket #2018: Increase resolution of length units in Rocket display

neilweinstock, 2023-02-03:
Right now measurements in Inches are given 3 decimal places.  This is a resolution of .001 inch, or .025 mm.

Metric measurements, on the other hand, are shown to a resolution of 1 mm (consistent if the chosen units are mm, cm, or m).  This is a resolution of... well, 1 mm.

Although 1mm seems more than adequate, it turns out that it is not enough to ensure that the displayed values of CP, diameter, and stability margin are consistent.  The numbers behind the scenes are all correct, but if you try to double-check your stability margin based on what you are shown in the rocket display, you will get an error, purely due to the round-off error in the displayed values.

There are at least three possible solutions to this.
1) Add an additional decimal point to the metric measurement display
2) Set display precision by significant digits (probably 4), rather than decimal places.  This would require a bit of heuristics, because we do not want to suddenly start displaying all numbers in scientific notation.  So I would only apply the limit to values that require decimal places to be displayed.  So 12345.56789 would be displayed as 12346.  But 45.6789 would be displayed as 45.67.  Something along those lines.
3) Don't worry about it, this isn't really a problem.

hcraigmiller, 2023-02-03:
Is this issue sufficiently related to Issue #830 such that one or the other should be closed?

neilweinstock, 2023-02-03:
I would say they are related but not the same.  The problem I'm pointing to here is the metric units, #830 regards inches.  And #2003 is not about these particular values.

Clearly there are plenty of places where precision can be revisited and/or improved.