openrocket/openrocket #2019: Component Analysis Dialog enhancements

JoePfeiffer, 2023-02-03:
A couple of things have come to mind trying to use the dialog to explain how OR is calculating drag on a recent rocket:

1.  when a row in the dialog is clicked, the corresponding component in the rocket design should be highlighted.
2. a fifth column should be added to the drag tab, giving the aggregate total Cd for each component.  The current fourth column, showing per instance total Cd, should be retained. The grand total Cd should be over this fifth column.

SiboVG, 2024-09-06:
> 2\. a fifth column should be added to the drag tab, giving the aggregate total Cd for each component.  The current fourth column, showing per instance total Cd, should be retained. The grand total Cd should be over this fifth column.

Should we blank out the per instance CD of the rocket?
[image]

Doesn't make much sense to me to sum up the instance CD's and compare the relative influence. WDYT @JoePfeiffer?

JoePfeiffer, 2024-09-06:
> Should we blank out the per instance CD of the rocket? 
> Doesn't make much sense to me to sum up the instance CD's and compare the relative influence. WDYT @JoePfeiffer?

In the particular case of the whole rocket it makes sense to either blank it out or show it as equal to the rocket's total (so it isn't the sum of the per-component CDs). Blanking it out is probably marginally better, so we don't have users expecting it to be the sum of the per-component CDs.

Thinking about it further, this is only one example of a component assembly, which in general are hard to deal with in a table. Suppose the rocket has three pods; we should show the per-pod CD, and then the sum of the pods. But now suppose the pods each have four fins; we should show the per-fin CD, but then is the sum of the fins 4 times or 12 times a single fin?

SiboVG, 2024-09-07:
> Thinking about it further, this is only one example of a component assembly, which in general are hard to deal with in a table. Suppose the rocket has three pods; we should show the per-pod CD, and [...]

Good point. I guess for pods and booster, we just use one complete instance CD (sum of all children instances). For a rocket, I would still blank out the result, since that's really the grand total of all CDs.

SiboVG, 2024-09-07:
> Good point. I guess for pods and booster, we just use one complete instance CD (sum of all children instances). For a rocket, I would still blank out the result, since that's really the grand total [...]

Wait, currently we don't even add pods and booster in the component analysis tables... (because the CD is just 0)

PR #2557 (2024-09-10): [#2019] Highlight selected component in CA dialog in rocket panel  & Add per-instance CD column to component analysis dialog
This PR fixes #2019. When selecting a row in the component analysis tables, the corresponding component is highlighted in the rocket panel. There is now also a column for per instance CD and total CD. In the per instance CD, the cell of the rocket is left blank and there are no relative contributions (the percentages).

https://github.com/user-attachments/assets/9ec575d8-c3af-4e11-a068-68a06bd0f70f
closed 2024-09-10 by SiboVG