openrocket/openrocket #2024: Get rid of option to save subset of simulation data

JoePfeiffer, 2023-02-06:
**Is your feature request related to a problem? Please describe.**
This is an option that was introduced when disks were much smaller (possibly at the beginning of the project). ork files, even with many simulations, just aren't big enough to worry about any more.

There appears to have been at least one case of a user who used this option to save a subset of the data, and then thought the data hadn't been generated.

**Describe the solution you'd like**
Just get rid of the "Every xx seconds" option in the file save dialog

**Describe alternatives you've considered**

**Additional context**
Add any other context or screenshots about the feature request here.

neilweinstock, 2023-02-06:
I would love to get rid of this option.  Every time I save a file and the dialog comes up I wonder why it is asking me that.

How big could a file get if all data were saved at max resolution?  Thinking for a high-flying rocket with a lot of simulations.

JoePfeiffer, 2023-02-06:
The .ork file from one run of an "orbital" design, which times out at 1200 seconds -- it hits a peak altitude of 1.29e7 ft -- is 73,563,451 bytes. That's pretty much a worst case (I came across the design because the user seemed to feel our chopping a sim off at 1200 seconds was a bug).

Remember the suggestion is just to not allow saving part of the data.  It leaves the option of saving no data at all available.

neilweinstock, 2023-02-06:
That is big. :)

We then need to have a sensible policy for what to do when the file is loaded.  I guess someone doing something like that would have "automatically update simulations" disabled, don't want to kick off the sims every time a file like that is loaded.  I guess in that case the sims would just be marked "out-of-date", and it would behave as normal according to that scenario.

I do like an "all-or-nothing" approach.  The middle option is just confusing in most cases.

JoePfeiffer, 2023-02-06:
The .ork has 9 identical (I think -- they at least all use the same motor configuration) simulations defined.  I wanted to present a *real* worst case of saving all of them, but life is just too short to run them all. They're several minutes each.

We should be showing sims as having been loaded from a file when that's the case.

PR #2070 (2023-02-26): Properly load simulation data from saved simulations
It turned out the old code never loaded sim summary data; the one row I thought had done so in the original bug report had actually been automatically re-run when I loaded the sim (which may be a bug itself, but if it is it isn't the same bug)

This PR fixes it by:
1.  Any time you've got flight data, load it (primary problem with the old code was it wouldn't load summary data if there was no detailed data)
2. Mark simulation as LOADED if came from a file.  Previously had logic based both on that and the status tag saved with the simulation. [... cut]
PR #2078 (2023-02-26): Eliminate option to save "some" simulation data
Save options now allow saving all sim data or only data from the sim summary table

Changes "Only primary figures" option name to "Only summary data" for clarity

Fixes #2024
closed 2023-02-26 by JoePfeiffer