openrocket/openrocket #2025: macOS 3D figure, unfinished, finished - use any view crashes if sharing desktop on Zoom

jimmiedave, 2023-02-09:
**Describe the bug**
When sharing OpenRocket on Zoom, opening an example rocket, OK'ing the Info dialog, selecting 3D Figure, 3D unfinished or 3D finished never brings up these views, but instead a "spinning pizza" cursor, locking OpenRocket and requiring a force-quit.

**To Reproduce**
Steps to reproduce the behavior:
1. Open OR
2. Option+[green maximize button] to maximize OR without entering Full-Screen mode
3. Share the desktop with Zoom on macOS, command+TAB screen back to OR
4. Load an example rocket. I tried several, but Pods -- powered with recovery deployment demonstrates this
5. Click "OK" to close the example info dialog
6. Select 3D figure, 3D unfinished, or 3D finished from the "View Type" menu on the design view
7. OR presents a spinning pizza cursor, locks up, must be force-quit.

**Expected behavior**
View switches to the selected one, and OR doesn't lock up.

**Screenshots**
Locked up. Still showing line drawing. Screenshots are n/a

**Desktop (please complete the following information):**
 - OS: macOS 10.15.7 on 2019 Intel Macbook Pro 16-inch on built-in screen, no external monitor
 - OpenRocket 22.02 production
 - Zoom 5.12.9 (13190)

**Additional context**
After quitting Zoom, without a restart of computer or OR, the behavior disappears.

Sorry to file a bug on day 1!

hcraigmiller, 2023-02-09:
> Sorry to file a bug on day 1!

Nothing to be sorry for, you win first prize. Our gratitude for the report.

jimmiedave, 2023-02-09:
Just had it happen again, this time in Photo Studio, but screen was not shared, though Zoom running in background, without a meeting going. Needs a little more testing.

SiboVG, 2023-02-09:
Could you go to the preferences in the general tab and click the "Reset all preferences" button? We've had a user some weeks ago complaining about PhotoStudio being really slow, and that fixed it for him.

jimmiedave, 2023-02-09:
I'll try that first. Just sat down to bang on this. Thought I'd also reboot in "safe mode" and see if something installed is getting in the way.

jimmiedave, 2023-02-09:
Still happening, no Zoom running now.

Reset preferences doesn't fix
Off-screen rendering doesn't fix
Rebooting to safe mode.

jimmiedave, 2023-02-09:
Solved: problem traced to a login item "Cinch" which hasn't been updated since 2020, and is apparently incompatible with macOS Mojave (I'm on Catalina). Removed this item and have not been able to repro since. Please remove this bad report.

SiboVG, 2023-02-09:
Glad to hear it's been resolved :)

closed 2023-02-09 by SiboVG