openrocket/openrocket #2026: Allow fins to be attached to tube fins

neilweinstock, 2023-02-09:
This seems like it would be straightforward to implement (simpler than pods-on-fins), and would allow designs like the Estes Super Neon without needing to fiddle with pods.
[image]

When we one day support ring fins, we will also want to allow fins to be attached to them, for the same reason.

JoePfeiffer, 2023-02-09:
The general idea of fins-on-fins -- including tube fins -- is definitely something to look at for next release. It ought to let us do a better job with the aerodynamics than our current ugly fins-on-pods hack (he said without actually having looked in to what's involved)

neilweinstock, 2023-02-09:
I was thinking that the fins-on-tube fins was a distinct special case vs. the general fins-on-fins problem.  We already know how to model a fin on a tube, and no new special parameters or hierarchy are required.  We "simply" need to allow it, and then model it just like a fin on a body tube.

Fins-on-fins (the normal kind), on the other hand, requires a considerable amount of new design work.  It seems like we could (and should) keep this as a separate item, and not let it get tangle up with the rest.