openrocket/openrocket #2028: Tumble recovery

JoePfeiffer, 2023-02-09:
**Is your feature request related to a problem? Please describe.**
OR currently can recognise some situations where a rocket stage is tumbling (eg boosters following separation), but has no way to specify a tumble recovery for a rocket.

**Describe the solution you'd like**
Add a "tumble" option to the recovery configuration.  Notice that at present the recovery configuration isn't even available if you don't have a recovery device defined.

hcraigmiller, 2023-02-09:
Would this feature solve the chute release issue, in whole or part (setting the chute release to tumble recovery)?

JoePfeiffer, 2023-02-09:
> Would this feature solve the chute release issue, in whole or part (setting the chute release to tumble recovery)?

I think we'll find that we won't do better on chute release than something similar to the small chute/big chute we're already recommending.