openrocket/openrocket #2032: NaN in simulation

JoePfeiffer, 2023-02-11:
Reported to openrocket-bugs@lists.sourceforge.net by pga2266670@aol.com.

Attached is the ork file. no way to attach the bug report.
I tried the other option but it was a royal PITA.
The problem appeared when I tried to do a flight simulation.
 
Sincerely
Paul Gagnon

I reran his .ork and confirmed issue under 22.02.
---------- Exception stack trace ----------
net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation
at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120)
at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:445)
at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:109)
at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750)
at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:847)
at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112)
at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:857)
at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:317)
at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:249)
at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117)
at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190)
at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235)
at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233)
at java.desktop/java.awt.Dialog.show(Dialog.java:1070)
at java.desktop/java.awt.Component.show(Component.java:1716)
at java.desktop/java.awt.Component.setVisible(Component.java:1663)
at java.desktop/java.awt.Window.setVisible(Window.java:1031)
at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1005)
at net.sf.openrocket.gui.main.SimulationPanel.runSimulation(SimulationPanel.java:361)
at net.sf.openrocket.gui.main.SimulationPanel.access$900(SimulationPanel.java:74)
at net.sf.openrocket.gui.main.SimulationPanel$RunSimulationAction.actionPerformed(SimulationPanel.java:573)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635)
at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
at java.desktop/java.awt.Component.processEvent(Component.java:6400)
at java.desktop/java.awt.Container.processEvent(Container.java:2263)
at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
[... 231 lines cut ...]
307 32.618 DEBUG (SimulationRunDialog.java:365) simulationStage boost: t=0.0 est=1.3132741671378534
308 32.618 DEBUG (SimulationRunDialog.java:455) Setting progress to 0 (real 0)
309 32.618 DEBUG (SimulationRunDialog.java:241) Progressbar value 0
310 32.620 ERROR (SwingExceptionHandler.java:119) An exception occurred during the simulation
> net.sf.openrocket.util.BugException: BUG: Simulation resulted in not-a-number (NaN) value, please report a bug, c=(NaN,NaN,NaN,w=NaN)
> at net.sf.openrocket.simulation.AbstractSimulationStepper.checkNaN(AbstractSimulationStepper.java:221)
> at net.sf.openrocket.simulation.RK4SimulationStepper.computeParameters(RK4SimulationStepper.java:296)
> at net.sf.openrocket.simulation.RK4SimulationStepper.step(RK4SimulationStepper.java:121)
> at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulateLoop(BasicEventSimulationEngine.java:178)
> at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulate(BasicEventSimulationEngine.java:119)
> at net.sf.openrocket.document.Simulation.simulate(Simulation.java:406)
> at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:55)
> at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:24)
> at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304)
> at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
> at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343)
> at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
> at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
> at java.base/java.lang.Thread.run(Thread.java:829)
311 32.620 INFO (SwingExceptionHandler.java:142) Running in EDT, showing dialog
312 32.620 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
313 35.706 USER (SwingExceptionHandler.java:220) User requested sending bug report
314 35.709 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------
[Space.ork.txt](https://github.com/openrocket/openrocket/files/10712279/Space.ork.txt)

hcraigmiller, 2023-02-11:
The problem is the phantom elliptical fin set. Remove that, and it simulates without error.

[image]
[image]

PEGagnon, 2023-02-11:
FANTASTIC !!! As I use this program I'm learning how much I'm liking it.
For my next trick I need I need to learn how to put a booster under it.
Your help in this matter is GREATLY appreciated. I'm Very pleased. I will speak highly of this product.
Paul Gagnon

PR #2082 (2023-03-10): Tell user about zero-area fins and keep them from crashing simulation with NaN error
Warn that fins with zero area will not affect aerodynamics
Keep fins with zero area from getting NaN errors

Fixes #2032
closed 2023-03-10 by JoePfeiffer