openrocket/openrocket #2036: [Bug] Parachute Location error after updating packed length

Rocketman-1969, 2023-02-11:
**Describe the bug**
The Location for the parachute icon is in the wrong place when you enter the packed length and the parachute location relative to the bottom of the parent component. 

**To Reproduce**
Create a parachute, change the packed length to 4 in, change the relative location to the bottom of the parent component, and change the location. 

**Expected behavior**
I would expect that the parachute icon would be at a reasonable distance from the bottom of the parent component 

**Screenshots**
[image]

**ORK File**
If the problem is triggered by a particular ORK file, please zip it and attach it to this issue.

**Platform**
 - OS: [e.g. Windows10]
 - OpenRocket version (e.g. 22-02)
 - Graphics card (Ninvida 

**Additional context**
Add any other context about the problem here.

SiboVG, 2023-02-11:
Can confirm if the "Auto" packed diameter checkbox is checked.

Rocketman-1969, 2023-02-11:
I unchecked it and it seemed to resolve the issue

PR #2037 (2023-02-12): [#2036] Fix incorrect position of parachute when auto radius
This PR fixes #2036. The issue was that some places use a direct call to `this.length` instead of using the `getLength()` method.

I never noticed it before, but the component CG of the parachute would not update when changing the auto radius (if the parachute suddenly halved in length, its CG should also update - which was previously not the case). This PR fixes that as well.
closed 2023-02-12 by SiboVG