openrocket/openrocket #2045: [Feature Request] Toolbar for PhotoStudio

SiboVG, 2023-02-12:
I think it would be useful to add a toolbar to PhotoStudio, to have quick access to the settings, and to make room for new features, such as exporting a PhotoStudio view (the camera icon on the screenshot below), and saving PhotoStudio view presets (see #2044).

[image]

neilweinstock, 2023-02-13:
I like the approach.

I presume preset views should be app-wide (vs. file-specific)?  How do you delete one?

Bigger question: what should be included in the preset view?  Normally I'm thinking of the stuff in the first settings tab, and also Lighting (from middle tab).  I will often choose a different background for different rockets but still want to use the same camera angles and such.  I'm not sure about this.

SiboVG, 2023-02-13:
> I presume preset views should be app-wide (vs. file-specific)? How do you delete one?

I was actually thinking file-specific, I don't see how you would want app-wide presets since it depends on each rocket design, right?

neilweinstock, 2023-02-13:
I don't know.  There are certain views that could work pretty well for all rockets, e.g. side view, rear view, 3/4 front or rear view.  Tedious to need to recreate all those views each time you create a new rocket.

SiboVG, 2023-02-13:
We can always start with file-specific views and if we want to add app-wide views, we can do so later on?

SiboVG, 2023-02-13:
I don't see that much use in app-specific presets, but I'll let you think about it.

neilweinstock, 2023-02-13:
Consider this.  For all my rockets, I create some "standard" Photo Studio renders beforehand to visualize.  If all the presets are file-specific, I'll need to re-enter them for each rocket.

The main rocket-specific settings I can think of are the background (although only sometimes), and the distance between camera and rocket, which obviously needs to be larger for larger rockets.

neilweinstock, 2023-03-03:
After further consideration I am reasonably well convinced that global presets are the more useful feature, at least initially.

Where things get a bit dicey is in the "distance" parameter, which will need to vary considerably from rocket to rocket.  An "auto" function on distance would be helpful there.