openrocket/openrocket #2051: [Bug] <insert your title>

JoePfeiffer, 2023-02-14:
**Describe the bug**
Quotes in .ork files cause a failure to load

**To Reproduce**
Attempt to open the attached file.  An error popup appears, stating the the XML is malformed.  The issue is the 11" long tube.

**Expected behavior**
File opens

**Additional context**
Note that a workaround is to load the problematic part from the parts library, then change a parameter and then change it back so the part name isn't saved.

[ork_test.zip](https://github.com/openrocket/openrocket/files/10727839/ork_test.zip)

closed 2023-02-16 by SiboVG

sjh114961, 2023-02-16:
The right thing to do is fix the ORC file with the error. Please add an issue to DCooks parts database.

SiboVG, 2023-02-16:
> The right thing to do is fix the ORC file with the error. Please add an issue to DCooks parts database.

Not really necessary, we can sanitize the XML ourselves (see my PR #2054). Dave shouldn't worry about this kind of stuff :)

PR #2054 (2023-02-16): [#2051] Sanitize some XML content
This PR fixes #2051 by sanitizing some of the XML content when saving a .ork file.

Note: it is not possible to open files already saved with quotes, it is only possible to avoid this issue from happening in the future by sanitizing the XML content before saving it.

Steps to reproduce:
1. Add a body tube, go to its parts library
2. Search for "38mm Motor Mount tube 11" and apply it to the body tube
3. Save the file and close it
4. Open it again