openrocket/openrocket #2059: [Feature Request] Combine "Inner Components" and "Mass Objects" groups in the "Add Components" button panel

neilweinstock, 2023-02-15:
There is no meaningful distinction between Inner Components and Mass Objects.  Two of the four mass objects (parachute and streamer) have functionality beyond just being mass objects, so I don't know why they're in that group in the first place.  And if those moved to Inner Components, then Shock Cord should really go with them.  That would leave only "Mass Object" in the "Mass Objects" group.

Merging the two groups into simply "Inner Components" makes the most sense to me.

SiboVG, 2023-02-24:
I'm not sure if I like how few component groups that would effect to. Maybe change "Mass components" to "Recovery component", containing only the parachute, streamer and shock cord, and move the mass component to the inner components?

The reason why I like the component groups is that you don't have to search through a large row of components to find the desired component. Just makes stuff a bit more organized.

neilweinstock, 2023-02-24:
Well, there are 3 recovery components and 1 mass object.  IMHO separating them out does not add any benefit.  Certainly the current labeling of "Mass Components" makes no sense, and if you want to label the three of them (parachute, streamer, shock cord) "Recovery components" and put Mass back into Inner Components I suppose that's OK but I still don't think it's super useful.

In the screenshot I posted to Discord where everything has been compacted, you can see that there just isn't that big of a group of inner components.  It is not hard to find anything,  In the current layout of the component buttons, things do get painfully spread out, but that is a separate problem to fix.

Do you really think in this mockup it would be hard to find anything?
[image]

SiboVG, 2023-02-24:
Well I'm not sure what to feel about the mock-up, the component buttons feel very cramped. Some vertical separation between the groups would help a little, but I don't know if I like it yet.

hcraigmiller, 2023-02-24:
I'm not sure that combining components should be done at this time. Not only is a complete revamp of the component icon area being discussed, there are also current issues pending that could add additional icons: (1) chute release; (2) freeform object; and potentially others.

Insofar as grouping is concerned, I prefer to keep an "inner" component and a "recovery" component group that would consist of the following:

- Inner Components: Inner Tube; Tube Coupler; Engine Block; Centering Ring; Bulkhead; Freeform Object.
- Recovery Components: Parachute; Chute Release; Streamer; Shock Cord; Mass Object (the "mass types" are primarily recovery in nature).

Combined, this would be a group of 11 icons, too many for my liking.

hcraigmiller, 2023-02-24:
I rather like moving the editing buttons, but I would prefer either of the following layouts:

[image]

I do somewhat prefer A over B.

neilweinstock, 2023-02-24:
Debating UI is truly the bestest and most funnest thing in the world. :)

> Well I'm not sure what to feel about the mock-up, the component buttons feel very cramped. 

Things could be spaced out a bit more if needed, it's just a discussion starter.  But I would argue that the existing button layout is _absurdly_ spread out, especially on large windows, and the button layout changes all the time depending on window size.  I'm not sure I can think of another app that does it that way.

For what it's worth, the Rocksim buttons:
[image]

Anyway, back to the topic of this feature request.  Craig wrote:

> I'm not sure that combining components should be done at this time. Not only is a complete revamp of the component icon area being discussed, there are also current issues pending that could add additional icons: (1) chute release; (2) freeform object; and potentially others.

This is a very reasonable argument for not doing this now.  After we add some additional buttons we can decide if a reorganization makes sense.  I feel like the current grouping (or at least the label) is sort of nonsensical, but no point in making the change until we have the bigger picture under control.

I'm very receptive to the idea of renaming the group "Recovery Components", though I'm not really convinced that "Mass Object" is most commonly one of those.  The #1 use for it has to be as nose weight.  However, "Recovery Components" is, on the whole, a much more appropriate name than what we have now.

Regarding moving the edit buttons: I forgot that Windows puts the tabs all the way on the left... given that, putting the buttons below seems reasonable.  I also prefer A over B.  But I wouldn't do this change until something is done with the Component button area.  This will just make that area even more ridiculously oversized.