openrocket/openrocket #2060: [Bug] Multi-level-wind plugin no longer works on 22.02

SiboVG, 2023-02-16:
**Describe the bug**
Reported by user u/maxjets on the r/rocketry Discord server: The multi-level-wind plugin no longer works on 22.02. Plugin link: [https://www.rocketryforum.com/threads/new-openrocket-plugin-to-allow-different-wind-speed-direction-at-different-altitudes.140619/](https://www.rocketryforum.com/threads/new-openrocket-plugin-to-allow-different-wind-speed-direction-at-different-altitudes.140619/).

It worked in OR 15.03.

SiboVG, 2023-02-16:
This may be more related to the plugin itself actually, I also noticed that thzero created a PR to update the plugin for the 22.02 betas: https://github.com/rocketsam2016/MultiLevelWind/pull/3.

However, the original owner of the plugin has not been active on GitHub since 2017...

This begs the question: should we take over the MultiLevelWind plugin? Or even integrate it natively in OR?

HappyCanuck, 2023-02-18:
I suggest it gets integrated into OR; I find it really useful for high-altitude flight simulations.
Also, it needs to be available as an option in the Preference/Simulation tab, so it can be used by default for all simulation cases.
Finally, it would be great to have the wind profiles as a separate file, similar to the .eng files, so we can use different cases easily.

SiboVG, 2023-10-10:
For those interested, you can download a 22.02 compatible version of the plugin [here](https://drive.google.com/file/d/1_RNlAHMZImoaG42EYPdkQ5mfaYSQj0lg/view?usp=drive_link).

JoePfeiffer, 2023-10-26:
In addition to making it native, I'd really like to find a wind model on line with a published API, so a user could just import current data for their launch location. We'd also want to be able to load and store the models, so they would be available if the user wasn't on line at the field.

HappyCanuck, 2023-12-20:
> In addition to making it native, I'd really like to find a wind model on line with a published API, so a user could just import current data for their launch location. We'd also want to be able to load and store the models, so they would be available if the user wasn't on line at the field.

Here is a site I found very useful: https://open-meteo.com/

JoePfeiffer, 2023-12-20:
Thank you! There's a stack of stuff first, but this will be very helpful when we get there.

PR #2559 (2024-09-25): [#2060, #2558] Implement multi-level wind input 
This PR fixes #2060 and fixes #2558, and I guess also #922 (except for the wind gradient request). Besides the already existing pink noise wind model, you can now also use a multi-level wind model for a simulation.

## UI
In the simulation config, there are 2 radio buttons to choose to either use the pink noise model, or the multi-level wind model.
[image]

Click `Add level` to create a new level, or `Remove level` to remove an existing level. If you want to show a visualization of the levels, you can click `Visualize levels`.
[image]

You can e [... cut]
closed 2024-09-25 by SiboVG