openrocket/openrocket #2064: Editing Graphics is broken

KTesh, 2023-02-22:
Editing the graphics is somehow broken.  I can open the chosen file, edit it, save it in the editor, but it doesn't update on the actual sim.  And when opening the graphic again, no changes have been made.

JoePfeiffer, 2023-02-22:
I thought this had already been reported (I know we talked about it), but can't find an issue for it.  It isn't #1991

KTesh, 2023-02-22:
We got through the problem of OR not opening the editor (MS Paint).

I pick a component I want to edit the appearance of.  Select the file I want to use.  Image appears on component.  Go to edit graphic, editor opens (Paint), I alter the graphic, save the file, and go back to OR.  No changes have been made.  Looks exactly as it did before I tried to edit it.  Try to edit it again... same results.

If I had to guess, OR is saving the file somewhere it shouldn't be.

My workaround is to save the altered graphic as a unique name, and then select that unique graphic "From file".  In the past, I could change a file dozens of times, save it, and see the results in OR immediately, without other graphics being created/named.  It's a serious Pain(t) in the ... (you know the rest).

neilweinstock, 2023-03-03:
I'm seeing the same behavior using Paint.net on Windows.  Haven't been able to verify yet on Mac, having other problems preventing me from doing so.