openrocket/openrocket #2067: [Bug] Save "only primary figures" doesn't save summary for all sims

JoePfeiffer, 2023-02-22:
**Describe the bug**
If we try to save a sim with the "only primary figures" option checked, not all simulation summaries are saved

**To Reproduce**
Start OR
Open "simple model rocket" example
Run simulations
Save to a new file with "only primary figures" option set
Close and reopen OR
Load the saved .ork
Observe summary table isn't fully populated
[image]

**Expected behavior**
All rows in summary table should have been populated

SiboVG, 2023-02-23:
May I also add that the term "Only primary **figures**" sounds very confusing? When I hear the word "figures", I think of graphs, not the numbers in the sim table. May be because I'm not a native speaker, but I had a very different notion of the intended behavior of that save function... (don't know what the behavior of "Only primary 'graphs'" would be, but in any case I find "figures" confusing)

JoePfeiffer, 2023-02-23:
> May I also add that the term "Only primary **figures**" sounds very confusing? When I hear the word "figures", I think of graphs, not the numbers in the sim table. May be because I'm not a native [...]

Yes.  I'm thinking of changing it to "summary data" while I'm at it.

PR #2070 (2023-02-26): Properly load simulation data from saved simulations
It turned out the old code never loaded sim summary data; the one row I thought had done so in the original bug report had actually been automatically re-run when I loaded the sim (which may be a bug itself, but if it is it isn't the same bug)

This PR fixes it by:
1.  Any time you've got flight data, load it (primary problem with the old code was it wouldn't load summary data if there was no detailed data)
2. Mark simulation as LOADED if came from a file.  Previously had logic based both on that and the status tag saved with the simulation. [... cut]
closed 2023-02-26 by JoePfeiffer