openrocket/openrocket #2076: [Feature Request] solid color and (especially) transparent backgrounds in photo studio

JoePfeiffer, 2023-02-25:
I've got a request from someone doing video for Spaceport America Cup to use our scripting interface to generate an animation of a rotating rocket.  The one part of it that I don't think OR does currently is he wants a solid or (ideally) transparent background from our photo studio.  That "ottabee" possible...

hcraigmiller, 2023-02-25:
Currently, Photo Studio will show a solid background by using the Edit > Environment > Sky Image > None option, then selecting the color.

[image]

I imagine that the RP just wants the image to save with a transparent background, but in OR a transparent background should probably look something like this:

[image]

JoePfeiffer, 2023-02-25:
Ah, I looked too quickly at what was already available.  But we still need to add transparent. I don't really care how it looks in our interface; being able to save with a transparent background is what's wanted here.

neilweinstock, 2023-03-03:
Is the idea for the script to:
a) position the rocket
b) save screenshot
c) go to (a)

That's interesting.  Please post some output if you're allowed.

In the meantime, if you use a slightly off-color background that doesn't coincide with anything in the rocket, it should be straightforward to remove it "in post".

JoePfeiffer, 2023-03-03:
It's from one of the people who works with livestreaming the Cup.
I don't think there's anything private in our conversation, so:

> It occurs to me there might be something interesting Livestream can do by having teams submit OR files and then automatically generating some standard animations to play in the Livestream

> The thing I would like would be a command line tool for Windows that looks like:
python3 script.py -input_file.ork -output_directoy -nFrames -backgroundColor
-input_file.ork
The input Open Rocket file, or a directory containing multiple Open Rocket files

> -output_directory
Each input ork file receives a subfolder in the output directory with the same name as the input file. Image filenames are the same as the Open Rocket .ork filename with _frame### appended

> -nFrames
Number of frames to render while rolling the rocket 360 degrees. For example if nFrames = 100, each image increments 3.6 more degrees of roll than the previous image.
The view for rendering is side view (Pitch = 0, Yaw = 0, View Az = 0, View Altitude = 0, FoV = 70, and maybe some math to set View Distance and Advance to fill the screen) with size 1920x1080 (1080p).

> -backgroundColor
RGB tuple that defines the solid background color for OR rendering. PNG with transparent background would be ideal, but looks like it isn't supported?

I remarked "green screen for the win!" as long as it isn't the Notre Dame team, of course...

neilweinstock, 2023-03-03:
Cool.

PR #2157 (2023-04-01): [#2076] Support transparent rendering and exporting of PS images
This PR fixes #2076 and adds support for transparent background rendering and exporting in PhotoStudio. A fully transparent color is just rendered as white.

Demo:

https://user-images.githubusercontent.com/11031519/228622619-3fb321ed-1a68-4e24-b1af-ac520e3b8783.mp4
closed 2023-04-01 by SiboVG