openrocket/openrocket #2077: [Feature Request] Add context menu ability for the simulation table to export all run simulation table rows to a CSV file

jppetrakis, 2023-02-25:
The ability to easily export the simulation data table to a csv for importation into a spreadsheet (eg. excel, google sheets, etc).  Once imported this can provide a portable, concise summary of the performance parameters of a rocket that  can be displayed on any given device (in particular a tablet at the launch site) or printed to a sheet of paper.

Being able to save the spreadsheet to google sheets and then referenced during a launch via a tablet or smartphone gives great flexibility to the user.

Considered directly printing the simulation table to a printer, pdf etc.  The approach of producing a csv file seems simpler.  I can see future expansion of this "command" to include different formats such as pdf or some other document type.

hcraigmiller, 2023-02-26:
See also, Issue #1525 "After that, I'd like to either be able to select rows and cells and copy them in a way that pastable into , or export the summary table as a delimited file."

JoePfeiffer, 2023-02-26:
#1525 probably should have been two feature requests in the first place -- the desire to export the table contents is pretty independent of the desire to configure the table.

hcraigmiller, 2023-03-09:
Resolved by way of PR #2087.

closed 2023-03-09 by hcraigmiller