openrocket/openrocket #2079: [Feature Request] Add option to show stability in cals and as a percentage

Ash2772, 2023-02-26:
I find my self frequently switching between stability calibers and the stability as a percentage.

Could there be an option in the units preferences to show cals and % at the same time in the rocket design tab?

[image]

neilweinstock, 2023-05-23:
I agree with this.  We should show both all the time, and remove the option.

Propose something like this:
[image]

SiboVG, 2023-05-23:
> In the Simulation Plot/Export, there is currently "Stability Margin Calibers" available as a Y axis type. I would then add "Stability Margin Percentage" as an addition type. I _think_ that's the only other place in the UI that stability appears.

I would make a new ticket for this to add units to the "Stability Margin" data type. I would prefer this over having a separate "Stability Margin Percentage" data type.

neilweinstock, 2023-05-23:
Yes, of course that is better.  I was working too fast and forgot that there was a units option for each Y axis type. I'll create the new issue.

PR #2220 (2023-05-29): [#2079] Display secondary stability unit
This PR fixes #2079 and adds the stability in percentage length in addition to the stability value in the currently selected stability unit. If the selected unit is already the percentage, or the stability value is N/A, the stability percentage is not included.

https://github.com/openrocket/openrocket/assets/11031519/0222cefd-01eb-49aa-ae48-0b6cb8399227
closed 2023-05-29 by SiboVG