openrocket/openrocket #2083: [Bug]  UnitsOfMeasure for materials in .orc files has no effect

dbcook, 2023-02-27:
Code analysis and functional testing shows that the UnitsOfMeasure attribute for <Material> has no effect in .orc files.

No matter what you specify, the UoM is always interpreted as the SI unit (kg/m3, kg/m2, kg/m).

Looking at some code history suggests this has been true for a very long time, perhaps since the beginning of OR.

To reproduce:  Change the UnitsOfMeasure on some <Material> entry in a .orc file in the parts database and observe that the computed mass of parts made from that material don't change.

Expected:  Units of measure will be interpreted as specified.

Note: Fixing this will break old versions of the parts database prior to 1.0.0.3

SiboVG, 2023-02-27:
Hi Dave

Thanks for the bug report, good catch! Looking at the code, the UoM is extracted from the .orc, but when generating the material, the UoM is left out of the party.

When you say "break old versions of the parts database", what do you mean exactly?

Cheers,
Sibo

To reproduce: Change the UnitsOfMeasure on some entry in a .orc file in the parts database and observe that the computed mass of parts made from that material don't change.

dbcook, 2023-02-27:
I must apologize, I originally discovered this years ago when I first went
through all the code to see how things worked, but later forgot that I
hadn’t filed an issue.

The fix will have to be issued with at least version 1.0.0.3 of the parts
database (just pushed last night) since prior to that time there were a
number of materials with the correct numeric SI value but the wrong UoM
tag.

Dave

SiboVG, 2023-04-10:
@dbcook, just so I get this right, does the density value of a material use the unit defined by the `UnitsOfMeasure` attribute? E.g. if I have this:
```
<Material UnitsOfMeasure="g/dm3">
  <Name>Test, bulk</Name>
  <Density>894.4</Density>
  <Type>BULK</Type>
</Material>
```

Then the density is 894.4 g/dm3? And if OR is configured to show bulk densities in g/cm3, then it will display as `Test, bulk (0.894.4 g/cm3)`?

(As you can tell, I'm not too familiar with how the material/component database should work.)

dbcook, 2023-04-10:
The density units aren’t used.  No matter what the units say, the value is
always parsed and used internally as kg/m3. Unless they have fixed it since
I filed the issue recently.

SiboVG, 2023-04-10:
> The density units aren’t used. No matter what the units say, the value is always parsed and used internally as kg/m3. Unless they have fixed it since I filed the issue recently.

It still works that way. I guess I wasn't clear in my original message, **_should_** the behavior be as stated above? That the density value is relative to the unit defined by UnitsOfMeasure, instead of always the SI unit?

dbcook, 2023-04-10:
Ahh now I understand 😁 Yes, if the units are going to be specified, they
should be observed when parsing.  There is a caveat about making sure that
the latest parts database is included when the fix is released, as the fix
will break all previous versions of my database prior to about a month ago
since there used to be many entries that had the UoM string at gm/cm3 but
with the correct SI value.

PR #2191 (2023-04-22): [#2083] Account for units of measure in material database loader
This PR fixes #2083 and accounts for the units of measure in the material object in a .orc file. The density parameter in the material XML object is now relative to the specified UnitsOfMeasure attribute of the material. E.g.:
```XML
<Material UnitsOfMeasure="g/dm3">
  <Name>Test, bulk</Name>
  <Density>894.4</Density>
  <Type>BULK</Type>
</Material>
```

Here, the density is `894.4 g/dm3`. If the bulk density units are set to `g/cm3` in OR, then the material will display as `Test, bulk (0.894 g/cm3)`.

Previously, the density va [... cut]
closed 2023-04-22 by SiboVG