openrocket/openrocket #2092: [Bug] Wacky simulation results with motors in pods

neilweinstock, 2023-03-02:
Running an unstable build of 22.02.

I took the "Simple Model Rocket" example and added some pods to it, with motors (was preparing a tutorial for TRF).  The sim results are crazy:
[image]

The Apogee results show the effect of the very early deployment on the -0 and -2 second delays of the pod motors.  But the deployment velocity is exactly the same for both, as is max velocity.  This makes absolutely no sense.

Here is a sim plot of the first one:
[image]

OK, this is weird... it shows a deployment at first motor burnout (about 1 second), which is the burnout of the A10... but then it shows *another* deployment at 5 seconds, which is the ejection charge of the B4-4.  But there is only one parachute in the rocket.  

And this is why it's showing a 10 mph deployment velocity... it's only showing the *second* deployment (which is not really a deployment at all), and at that point the rocket is descending peacefully on the parachute that was deployed at 1s.  You can see the second deployment isn't real because the descent velocity doesn't change.

So basically there is all sorts of confusion in here in how the pod motors are being treated.

Here's the ORK file.
[pod_motors_broken.ork.zip](https://github.com/openrocket/openrocket/files/10867109/pod_motors_broken.ork.zip)

hcraigmiller, 2023-03-02:
To clarify the observations of @neilweinstock, The sustainer recovery device trigger is (but should not be) linked to the pod motor and the pod motor burnout and ejection charge are both being treated as a recovery device triggering event. If the sustainer recovery device triggering event is set to a parameter that is the same for both motors (such as launch + X sec) then there is only one recovery device trigger.

Until this is fixed, the hack is to set the recovery device deployment to a recovery device triggering event that is the same for both the sustainer and pod motor(s).

neilweinstock, 2023-03-02:
I updated the title so it more accurately reflects the problem, now that it's pretty clear what it is.