openrocket/openrocket #2100: Can't Delete Unused Appearance files...

KTesh, 2023-03-06:
Since I can't successfully edit an appearance file anymore, I have some of them that are 11 files deep now, and the accidental carryover from a copied part...  I'd love to be able to select those extra files... and delete them.

[image]

hcraigmiller, 2023-03-09:
In clarification. if there are unused decal files, then, when the design is saved and reopened, the unused files are no longer shown in the decal list, but the unused files are still located in the decal directory.

KTesh, 2023-03-09:
I'd just rather be able to edit the old files... and not have to make multiple files because I missed this or that (or needed to change something).

The old way worked *SO​* much easier.  Create file, save file.  In OR open file, edit file, save... It works...

Now it's create file, save file... In OR open file, edit file, save... no change...  Save file with different name...  In OR open new file, and if something is wrong... Rinse... Repeat (ad nauseum).  Right side fins are the absolute worst.

JoePfeiffer, 2023-03-09:
Yes, this is a regression which we need to fix.