openrocket/openrocket #2104: [Feature Request] Dynamically update International Standard Atmosphere temperature and pressure text

SiboVG, 2023-03-07:
When ISA conditions are selected, it would be useful if the pressure and temperature change when the altitude changes, so users don't think the temperature and pressure is always stuck at 15 °C and 1013 mbar.

[image]

SiboVG, 2023-03-07:
Well I guess the text boxes don't even display the ISA conditions, they display what was typed in before. I think it's better to display the ISA values there regardless of what was typed in the boxes before.

JoePfeiffer, 2023-03-07:
I agree completely -- see #1217 :)

SiboVG, 2023-03-07:
> I agree completely -- see #1217 :)

Ahh, I searched the open issues, so I didn't find it. Well I have to disagree with my March 15th 2022 opinion, I actually like your #1217 proposition Joe 🙃.

PR #2133 (2023-03-24): [#2104] Display ISA values in temp & pressure fields
This PR fixes #2104. 

https://user-images.githubusercontent.com/11031519/227043260-b00937c2-8abb-47ea-a215-8ce09f29f38e.mp4
closed 2023-03-24 by SiboVG