openrocket/openrocket #2106: We really need to be able to choose a background color in the build window...

KTesh, 2023-03-09:
It's hard to see white on white with the shine at 100%... (How many white fins do I have there?  Are any missing?)

[image]

And the photo studio tool, which is nice, doesn't help when I'm trying to show off my work the way I want it to be seen...

[image]

I'd really love to see a background colour change option.

Thanks!

hcraigmiller, 2023-03-09:
> For the 3D view, you could have basic settings such as the background color and FoV

I have experienced the same desire to have a different 3D background color. I like your proposed placement of the pop-up button.

neilweinstock, 2023-03-25:
Keep in mind that this will require us to dynamically select the color of text that is placed in the viewing area, or else we'll have to fence in all the text with panels or some such.

SiboVG, 2023-03-29:
> Keep in mind that this will require us to dynamically select the color of text that is placed in the viewing area, or else we'll have to fence in all the text with panels or some such.

Good point. What would a good rule be? I guess the easiest would be to use the inverse of the background color for the text color (and then do a special inverse for the dark blue text on the bottom-left. But maybe something better is to switch between black text and white text, depending on the background color (brightness), like you have on security cameras?

neilweinstock, 2023-03-29:
I think choosing black or white is best.  Inverse color is not reliable.

SiboVG, 2023-03-29:
> I think choosing black or white is best. Inverse color is not reliable.

Agreed.

neilweinstock, 2023-03-29:
[image]

In the above image top and bottom text is the HTML color of the box.  The text in the middle is the luminance (.299R + .587G + .114B).  The black lines are the cutoff where we change from white to black, IMHO.

I find that white looks better ("pops") even when luminance is above the midpoint (128).  I would put the cutoff at around 140: below, use white text, above, use black text.

KTesh, 2023-03-29:
I think small windows for the data might be better (with an option of say white on black or black on white text).

[image]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F243872%2F228560283-f583e4fc-85c3-4759-8fda-ad69af1cb606.png&data=05%7C01%7C%7C05e3beaf8a214640b41f08db305d3a17%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638156949177490127%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=wfYUkQJS%2F6U5om%2FvQfHdDK93GO4dafAcZVcH0%2FIwGD8%3D&reserved=0>

neilweinstock, 2023-03-29:
> I think small windows for the data might be better (with an option of say white on black or black on white text).

I think that's probably going to look ugly.  

However, I just remembered that the warning text is red, and that's not going to be possible on arbitrary backgrounds, and the CP and CG symbols are placed next to the values in the upper right, and it's not going to be able to preserve their color either against an arbitrary background.

All this is why I was not to eager to jump on this particular feature request.  Gets messy.

KTesh, 2023-03-30:
The goal is to make the rockets look better...  I'm not so worried about the text.

I think that despite the ugliness of a text box it's a simpler (and thus more elegant) solution to the problem.  If the end user wants to change the text color, they can do that on their own, using their own graphics editor.

Perhaps though, there could be a sampler that shows alternate colors for the text against the background color, with the RGB values displayed for the user to select.