openrocket/openrocket #2107: Tube decal should not be applied to the edge

neilweinstock, 2023-03-09:
For tubes (Body or Inner), OR22.02 is applying the decal image, if any, to the ends of the tube.  As far as I can tell, OR15.03 did not do this.

Here is a comparison of 15.03 (left) and 22.02 (right), showing the effects of this (ORK file is attached):
[image]

Until something more intelligent can be done, only the color of the tube should be applied to the ends, not the decal.

[tube-test.zip](https://github.com/openrocket/openrocket/files/10932119/tube-test.zip)

hcraigmiller, 2023-03-09:
In clarification, 15.03 did not allow separate inner and outer appearance settings, which are now allowed, including setting the edge to either the inner or outer color. 

[image]

My concern, and where I think the bug is, is that the edge has any decal coverage at all in this situation:

[image]

neilweinstock, 2023-03-09:
> My concern, and where I think the bug is, is that the edge has any decal coverage at all in this situation:

Correct.

PR #2124 (2023-03-24): [#2107] Don't render decals on component edges
This PR fixes #2107. Decals are now not rendered on component edges anymore.
closed 2023-03-24 by JoePfeiffer